microsoft/Azure_Kinect_ROS_Driver #36: Lower the latency of IMU samples.

zchenpds, 2019-07-30:
**Is your feature request related to a problem? Please describe.**
The current implementation of [`void K4AROSDevice::imuPublisherThread()`](https://github.com/microsoft/Azure_Kinect_ROS_Driver/blob/1c6e77422f32306fe2be06c486ae66d75da4d21f/src/k4a_ros_device.cpp#L831-L861) causes a lot of IMU samples to be buffered and dropped because the call to `k4a_device_.get_imu_sample()` is not fast enough. (The IMU samples are generated at 1600 Hz but are comsumed at 300 Hz). As a consequence, the latency of the IMU samples being published is low at the beginning but could reach up to 500 ms very quickly. This is even much larger than the latency of the image captures.

> [DEBUG] [1564506266.837885822]: Depth image timestamp lags ros::Time::now() by
>                 93.1763 ms. The lag ranges from 85.5594ms to 149.853ms.
> [DEBUG] [1564506266.838092553]: Color image timestamp lags ros::Time::now() by
>                 93.5138 ms. The lag ranges from 85.6315ms to 156.59ms.
> [DEBUG] [1564506266.838987116]: IMU timestamp lags ros::Time::now() by
>                 507.203 ms. The lag ranges from 4.47381ms to 509.671ms.
> [DEBUG] [1564506266.842241953]: IMU timestamp lags ros::Time::now() by
>                 505.639 ms. The lag ranges from 4.47381ms to 509.671ms.

**Describe the solution you'd like**
We might want consider deleting [`loop_rate.sleep();`](https://github.com/microsoft/Azure_Kinect_ROS_Driver/blob/1c6e77422f32306fe2be06c486ae66d75da4d21f/src/k4a_ros_device.cpp#L860). By doing this, most of the time spent in this thread will be on waiting for `get_imu_sample()` to return as this function would block until the next sample is available. This would minimize the latency of published IMU messages. If a lower sampling frequency is needed, we could downsample it for publication.

Once the change is applied, the latency would look like this.
> [DEBUG] [1564509165.887144542]: IMU timestamp lags ros::Time::now() by
>                0.800075 ms. The lag ranges from 0.406712ms to 5.679ms.
> [DEBUG] [1564509165.891448754]: Depth image timestamp lags ros::Time::now() by
>                 87.1476 ms. The lag ranges from 87.1476ms to 107.465ms.
> [DEBUG] [1564509165.891636434]: Color image timestamp lags ros::Time::now() by
>                 87.4763 ms. The lag ranges from 87.4763ms to 107.699ms.
> [DEBUG] [1564509165.891814039]: IMU timestamp lags ros::Time::now() by
                4.82506 ms. The lag ranges from 0.406712ms to 5.679ms.

**Describe alternatives you've considered**
(None)

**Additional context**
(None)

skalldri, 2019-08-10:
Where do you see documentation / how did you determine that the IMU samples are generated at 1600Hz? The [K4A documentation](https://docs.microsoft.com/en-us/azure/Kinect-dk/hardware-specification#motion-sensor-imu) indicates that they're transmitted to the host PC at 208 Hz. 

Are they being transmitted in batches?

zchenpds, 2019-08-11:
It does look like they are transmitted in batches.

I have another program that calls [`k4a_device_get_imu_sample()`](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/group___functions_ga8e5913b3bb94a453c7143bbd6e399a0e.html) with `timeout_in_ms=1000` and logs the data to a csv file. This is what I got: [data_20190810_20_28_00_imu0.xlsx](https://github.com/microsoft/Azure_Kinect_ROS_Driver/files/3489417/data_20190810_20_28_00_imu0.xlsx) (`win_ts` is obtained by calling `GetTickCount64()` after `k4a_device_get_imu_sample()` returns). Apparently we can see the IMU samples are generated at 1.6kHz.

skalldri, 2019-08-11:
Ahh... I've spoken to @xthexder, who wrote the K4A playback / record API, and he has similar observations about the IMU data stream. He suggests that we use the [method used in the K4A record utility](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/f0a1229e243101b4fbdd5f8f0d6581a4fb3d8b6c/tools/k4arecorder/recorder.cpp#L198-L218) to acquire the IMU samples in batches.

PR #60 (2019-08-23): Fix IMU publish rate

## Fixes #36 

### Description of the changes:
- Read IMU samples from the device in a loop (like the K4A recorder does) to fully drain the queue of samples before sleeping

### Before submitting a Pull Request:
- [x] I reviewed [CONTRIBUTING.md](https://github.com/microsoft/Azure_Kinect_ROS_Driver/blob/master/CONTRIBUTING.md)
- [x] I [built my changes](https://github.com/microsoft/Azure_Kinect_ROS_Driver/blob/master/docs/building.md) locally
- [x] I tested my changes with a device

### I tested changes on: 
- [ ] Windows
- [x] Linux
closed 2019-08-23 by skalldri