microsoft/Azure_Kinect_ROS_Driver #42: driver.launch fails if depth stream is disabled

shuntaraw, 2019-08-13:
**Describe the bug**
driver.launch fails with errors if depth_enabled is set to false.

**To Reproduce**
1. Run: roslaunch azure_kinect_ros_driver driver.launch depth_enabled:=false point_cloud:=false rgb_point_cloud:=false
2. ROS node terminates with errors.

**Expected behavior**
ROS node runs without error and publishes RGB images and IMU data (and others) 

**Logs**
[2019-08-12 17:53:28.142] [error] [t=3742] /home/vsts/work/1/s/extern/Azure-Kinect-Sensor-SDK/src/transformation/transformation.c (84): transformation_possible(). Expect depth camera is running to perform transformation.
[2019-08-12 17:53:28.142] [error] [t=3742] /home/vsts/work/1/s/extern/Azure-Kinect-Sensor-SDK/src/transformation/transformation.c (101): transformation_possible(calibration, source_camera) returned failure in transformation_3d_to_3d()
[2019-08-12 17:53:28.142] [error] [t=3742] /home/vsts/work/1/s/extern/Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (919): transformation_3d_to_3d(calibration, source_point3d_mm->v, source_camera, target_camera, target_point3d_mm->v) returned failure in k4a_calibration_3d_to_3d()
terminate called after throwing an instance of 'k4a::error'
  what():  Calibration contained invalid transformation parameters!

**Desktop (please complete the following information):**
 - OS: Ubuntu 
 - Version: 18.04

zchenpds, 2019-08-15:
The cause is that both `K4ACalibrationTransformData::publishRgbToBaseTf()` and `K4ACalibrationTransformData::publishImuToBaseTf()` require the depth camera to be enabled because of these SDK API calls
https://github.com/microsoft/Azure_Kinect_ROS_Driver/blob/99fa52c6caa1a3db8b167b57267dd5891e1159f3/src/k4a_calibration_transform_data.cpp#L171-L175
https://github.com/microsoft/Azure_Kinect_ROS_Driver/blob/99fa52c6caa1a3db8b167b57267dd5891e1159f3/src/k4a_calibration_transform_data.cpp#L223-L226

I don't understand why calculating these TFs should involve the depth_camera_link. But they do look like they are static TFs so a quick fix would be to hardcode the TFs without using `convert_3d_to_3d()`?

skalldri, 2019-08-16:
Fixing this will probably require a small rewrite of the k4a_calibration_transform class.

The Azure Kinect's calibration data is relative to the sensors. There is actually no transform between the mechanical housing and any of the sensor co-ordinate frames, just between the sensors themselves.

To get around this, I made the `camera_base` frame the center of the visor, and used the mechanical specifications of the Azure Kinect to get the offset of the depth camera from the center of the visor. This is the static TF published by `K4ACalibrationTransformData::publishDepthToBaseTf()`. 

Once I had the static offset between the depth camera and the mechanical housing, I made all the other TF's relative to the depth camera so I could easily consume the K4A calibration data.

Unfortunately, if the depth camera isn't enabled, it's calibration isn't loaded, so you run into this issue.

I'll look into the way calibration data is published to see if I can work around this.

closed 2019-10-14 by ooeygui