microsoft/Azure_Kinect_ROS_Driver #44: P matrix in CameraInfo is incorrect

shuntaraw, 2019-08-13:
**Describe the bug**
P (projection matrix) in sensor_msgs/CameraInfo message should be defined for rectified images and not for raw images by convention (see the link below). Currently left 3x3 matrix of P is identical to K (intrinsic matrix) even if D (distortion) is nonzero.
http://docs.ros.org/melodic/api/sensor_msgs/html/msg/CameraInfo.html

**To Reproduce**
Steps to reproduce the behavior:
1. Run ROS node: 
$ roslaunch azure_kinect_ros_driver driver.launch
2. Subscribe to rgb/camera_info: 
$ rostopic echo -n1 rgb/camera_info

**Expected behavior**
left 3x3 matrix of P is different from K unless D is zero.

**Logs**
$ rostopic echo -n1 /kinect0/rgb/camera_info
header: 
  seq: 233
  stamp: 
    secs: 1565665931
    nsecs: 268071349
  frame_id: "kinect0_rgb_camera_link"
height: 720
width: 1280
distortion_model: "rational_polynomial"
D: [0.7799050211906433, -2.9802768230438232, 0.0007482637884095311, 5.6622509873704985e-05, 1.6997523307800293, 0.6572195887565613, -2.811908721923828, 1.6302372217178345]
K: [604.8837280273438, 0.0, 636.9334106445312, 0.0, 604.9359130859375, 367.4653625488281, 0.0, 0.0, 1.0]
R: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
P: [604.8837280273438, 0.0, 636.9334106445312, 0.0, 0.0, 604.9359130859375, 367.4653625488281, 0.0, 0.0, 0.0, 1.0, 0.0]
binning_x: 0
binning_y: 0
roi: 
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: False
---

**Desktop (please complete the following information):**
 - OS: Ubuntu
 - Version 18.04

shuntaraw, 2019-08-13:
I misinterpreted the definition of CameraInfo. The current implementation looks good.

closed 2019-08-13 by shuntaraw

Guptajakala, 2019-12-18:
@shuntaraw 
Why is this correct? I have the same understanding as yours