microsoft/Azure_Kinect_ROS_Driver #45: Integrate the body tracking feature to ROS

lesterlo, 2019-08-14:
**Is your feature request related to a problem? Please describe.**
No, just a feature request

**Describe the solution you'd like**
Could we integrate the body tracking data to ROS system which can display the joint data on RViz and utilize the joint data through on difference ROS module?

Also, I want to use the ROS system to record the body movement data.

**Describe alternatives you've considered**
No

**Additional context**
No

RoseFlunder, 2019-08-15:
I think they are already working on it because they included a modified k4a.hpp in their repo so that the C++ sensor API can be used with the C-API from the body tracking SDK.

However, I needed this myself and forked this repo to integrate bodytracking and other things.
If you need it fast too, you can take a look at this branch of my repo:
https://github.com/RoseFlunder/Azure_Kinect_ROS_Driver/tree/feat/body-tracking-multi-kinect-2

I will update the usage.md file with the additional parameters I added.

skalldri, 2019-08-15:
@RoseFlunder : are you interested in merging that code back into this mainline repo? You've got lots of great work there that resolves some of the open issues we have here (body tracking, playback support).

RoseFlunder, 2019-08-15:
Sure, I would be glad to help.
I just need some time to seperate the things into single pull requests I guess.

skalldri, 2019-08-15:
I'm also happy to take on the task of merging your branch back into the mainline repo if you want. I can put you on as a reviewer when I've got something put together.

RoseFlunder, 2019-08-15:
@skalldri 
I already made a pull request for the playback functionality which you can review and test.
I used the C-API because the C++ API is not yet public in the Sensor SDK 1.1.0

Could be rewritten to the C++ API at a later point in time when 1.2.0 is released but shouldn't be a lot of work.

lesterlo, 2019-08-16:
@RoseFlunder 
Thanks, I will try your repo first.

skalldri, 2019-10-01:
This was previously fixed.

closed 2019-10-01 by skalldri