microsoft/Azure_Kinect_ROS_Driver #56: RGB Pointcloud slows down the publishing rate a lot

RoseFlunder, 2019-08-22:
**Describe the bug**
When enabling the rgb_point_cloud the performance of the node degrades a lot. Without the colorized pointcloud the node can maintain a steady 30 hz publishing rate when using 30 FPS with the kinect.
With the rgb_point_cloud enabled the publishing rate decreases to like 10-12 Hz.
When visualizing the point cloud in RViz the slow publishing rate leds to a laggy view.
RViz itself maintains high enough FPS displaying the pointcloud but the data comes in too slow.
When using the k4aviewer there is no slowdown in colorized point cloud view.

**To Reproduce**
Steps to reproduce the behavior:
1. Start the node via the driver.launch and use fps:=30, depth_mode:=NFOV_UNBINNED, rgb_point_cloud:=false
2. Call "rostopic hz points2" in another terminal
3. Restart the node with rgb_point_cloud:=true
4. Call "rostopic hz points2" again and compare the rate with step 2

**Expected behavior**
Maintain the 30 FPS rate for publishing the rgb pointcloud on machines that can easily view the colorized point cloud at 30 FPS in the k4aviewer.

**Desktop (please complete the following information):**
 - OS: Ubuntu 18.04

skalldri, 2019-08-23:
Confirmed this is occurring on my ROS Melodic, Ubuntu 18.04 machine. I suspect the main image processing loop is running too slowly on the CPU, since the K4A Viewer does the RGB point cloud math on the GPU.

I'll investigate ways I could do this faster, potentially using SIMD instructions.

skalldri, 2019-08-23:
Confirmed, the image processing loop take between 40-70ms when the RGB point cloud is enabled. I'll look for ways to potentially speed it up.

RoseFlunder, 2019-08-26:
@skalldri
Is a GPU accelerated version planned?
In the usage.md this is stated but there isn't any GPU acceleration in the currently used SDK methods I think?

> Using the point cloud functions in this node will provide GPU-accelerated results, but the same quality point clouds can be produced using standard ROS tools.

Today I saw this example in the sensor sdk repository:
[Fast Pointcloud Example](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/tree/develop/examples/fastpointcloud)
Would this version already be faster even on CPU or is it comparable?
I guess this algorithm is implemented on the GPU for the k4aviewer isn't it?

RoseFlunder, 2019-08-27:
Thanks for the detailed answer :) I am looking forward to it because thats one of our main blocking points at the moment in our project.

RoseFlunder, 2019-09-12:
@skalldri
So like I already mentioned in another issue I did some time measuring.
In this gist I just pasted the modified part of the rgb point cloud method together with the output.
I don't know if ROS Time is very accurate we can clearly see that the SDKs methods are fine performance wise.
Whats really dragging down is the loop that creates the message:
https://gist.github.com/RoseFlunder/0cca13211022049ae7abc98ea600a7de

skalldri, 2019-10-10:
Going to start having a look at this.

skalldri, 2019-10-22:
Well, Valgrind seems to think that using `PointCloud2Iterator` is very expensive:

[image]

Combined, `PointCloud2Iterator::operator++` and `PointCloud2Iterator::operator*` account for ~20% of the time spent in `getRgbPointCloud()`! While not confirmed, this makes me think that we could see significant speedups if we do this the old fashioned way.

I'll see if I can build a version of the point cloud functions that doesn't use the iterator.

skalldri, 2019-10-22:
It's good to see that this is using `memset_avx2_erms`: that's probably the compiler optimizing the PointCloud2 alloc.

emmanuel-senft, 2020-01-10:
Hi, I am also affected by this bug, and I was wondering if there had been any progress on that side?

anastasiabolotnikova, 2020-04-13:
Hello, glad to know that this issue is getting attention and already started to be investigated. Thanks @RoseFlunder @skalldri  for sharing your findings! 

This is indeed a limitation of the current driver implementation that should be addressed to speed up the loop cycle. Both `fillColorPointCloud` and `fillPointCloud` could be much faster if going over the entire point cloud sequentially point by point is somehow avoided.

Three questions that I would like to ask regarding this issue:

1. Is there no way of converting 3D points coordinates data from K4A SDK structure (e.g. buffer) into [`sensor_msgs::PointCloud2.data`](https://docs.ros.org/melodic/api/sensor_msgs/html/msg/PointCloud2.html) binary blob using `memcpy`? I suppose that would be the fastest way to deal with it on machines without GPU?

1. Why `new PointCloud2` is being created and resized on every cycle? I suppose it only needs to be created and set to correct size once before the while loop of `framePublisherThread`?  

1. A parallel implementation of `fillColorPointCloud` and `fillPointCloud` for machines with GPU would be great! Especially if there is no way to copy points from SDK structure into ROS message  using `memcpy`. Could you share with us any information regarding the plans of this implementation?

Thank you in advance for the reply!

ooeygui, 2020-04-13:
Hi All,
We Azure Kinect to address a few performance related issues, as well as Migrating to ROS2 on our backlog, but likely won't be able to start working on it until the summer.

YoshuaNava, 2020-10-19:
@skalldri I'm also having issues with the PointCloud2Iterator, although I'm not a ser of Kinect Azure. Do you still plan to optimize its use? I would be happy to collaborate and find out better ways to use it, or come up with improvements.

ooeygui, 2020-10-19:
Hi @YoshuaNava ,
This work is still on the backlog.

The code currently does this work linearly in C++ https://github.com/microsoft/Azure_Kinect_ROS_Driver/blob/c8d4cb7f1d43326879f926294fb1aa4f299c159e/src/k4a_ros_device.cpp#L586.

We are looking at Cuda for this codepath when available.

YoshuaNava, 2020-10-27:
@ooeygui Thank you for your prompt answer.

I see. I have continued using the iterator. Until now it allowed me to speed up an old raw deserialization method by 3-5x, which is a good gain with considerable low effort.

I'll share any insights found while I work on this.

ibrachahine, 2022-03-11:
Hi, I am prototyping for an application requiring the depth topic on ROS and it seems to be very slow at 1Hz. Is there any recent updates on this issue? 

Thanks

ooeygui, 2022-03-11:
@ibrachahine Thank you for your interest and ask here. We have not invested in fixing this bug, but would accept a pull request. Have you tried workign with parameters on the node to optimize for your hardware and scenario?