microsoft/Azure_Kinect_ROS_Driver #57: Kinect timestamps and ros::Time::now() drift appart

skalldri, 2019-08-22:
**Describe the bug**
Over time, ROS Time and Kinect time appear to drift appart. This may impact the timestamping solution implemented in #21 .

**To Reproduce**
Leave the ROS node running overnight. Observe that timestamps drift significantly over long periods of time.

**Expected behavior**
Timestamps should remain relatively consistent over long periods of time.

**Logs**
Please provide relevant logs to help diagnose your issue. To assist in debugging, please enable *info* level logging of the Azure Kinect Sensor SDK and attach any logs you have to this issue.
See [k4atypes.h](https://github.com/Microsoft/Azure-Kinect-Sensor-SDK/blob/feecae0456511ac734287571b101d10fd7292673/include/k4a/k4atypes.h#L184) for how to enable logs.

Please also enable *DEBUG* level logging of the ROS node.
See [roscpp/Overview/Logging](http://wiki.ros.org/roscpp/Overview/Logging) for how to enable ROS DEBUG logs.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. Windows 10, Ubuntu]
 - Version [e.g. Version 1903, 18.04]

**Additional context**
Add any other context about the problem here.

RoseFlunder, 2019-08-23:
The sensor sdk 1.2.0 is released now, tho the debian/ubuntu package comes next week.
With the new version we have new API to get a image system timestamp instead of a device timestamp calculated from the SDK:
[https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/608](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/608)

But looks that they didn't provide such a method for the IMU sample unfortunately.

Maybe we can start to rewrite this logic with new API.
Use the new API for image timestamps.
Use the current approach only for IMU. So this "startTime" is only used by the IMU thread. No potential race condition, no need to protect access to start_time_ variable.

helenol, 2019-10-14:
@RoseFlunder Great about the new features provided in the SDK! However, there's a big potential problems with your proposed solution: the IMU and image time-stamps will then drift apart, negating the whole purpose of the of having an IMU on the same hardware clock as the image exposure.
I think a better solution would be to provide access to the mapping between the system and device time through the SDK, not just for a specific image.
A better interim solution would be to use some standard techniques to filter the device timestamps over time, such as: https://github.com/ethz-asl/cuckoo_time_translator

RoseFlunder, 2019-10-15:
@helenol
I understand. I also learned that the system timestamp from the SDK is not a mapping based on the device stamp.
Instead its defined otherwise:
https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/include/k4a/k4a.h#L899

> The system timestamp is captured at the moment host PC finishes receiving the image.

I guess we don't want this time right?

The device timestamp on the other hand is defined as follows:
https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/include/k4a/k4a.h#L873
> Returns the device timestamp of the image, as captured by the hardware. Timestamps are recorded by the device and represent the mid-point of exposure. They may be used for relative comparison, but their absolute value has no defined meaning.

bnjff, 2019-11-12:
The translator would be very useful. In our application, we need to sync the images with the proximity sensor triggering. With the current implementation, the camera's timestamps are lagging by a few hundred ms from the beginning (e.g. not just drifting over time), and the images cannot be synced correctly by message_filters. Since we're not using IMU, we solved it by reverting to the system time-based timestamps.

Brent-A, 2019-11-12:
FYI, for @wes-b who works on the timestamps in the SDK.

helenol, 2020-01-17:
Update from some triage... Doing tests on linux between device (AKDK hardware) timestamps, system (usb bus arrival afaik) timestamps, and the constant offset from start that we currently use.
I seem to experience a drift of about 1 ms/minute between device timestamp and "real" timestamp.

Did this by comparing image system timestamp + monotonic to realtime clock offset (evaluated at that iteration) vs. device timestamp + start_time from ROS.
The difference between the two grows steadily at about 1 ms per minute or so on my machine, which frankly is substantial.

Just need to do a portable between Windows and Linux solution for getting the monotonic-to-realtime clock conversion (I guess this is partially done in k4a with QueryPerformanceCounter(), and trivial in linux) and hopefully this should solve some of the problems.

Doing this approach should also solve the huge timestamp lagging issues @bnjff mentioned.

PR #108 (2020-01-20): Fix timestamp drift over time

## Fixes #
Resolves #57 

### Description of the changes:
- **Previous behavior**
  - Compute a fixed "start time" that converts between device and ROS wall time at the start of the execution, using a hard-coded "delay" time between either the IMU or image time.
  - In experiments, this time drifted away from what's reported as the system time by k4a (which is monotonic clock time at image arrival) by at least 1 ms/minute, though likely more.
- **New behavior**
  - Keep an evolving device to wall time estimate
  - Initialize estimate as before: comparing device a [... cut]
closed 2020-01-20 by ooeygui