microsoft/Azure_Kinect_ROS_Driver #59: Update Sensor SDK version to 1.2.0

RoseFlunder, 2019-08-23:
**Is your feature request related to a problem? Please describe.**
With the release of Azure Kinect Sensor SDK 1.2.0 we should switch to this new version.
[https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/608](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/608)

The MSI Installer for Windows as well as NuGet Packages are already released.
The debian package for Ubuntu follows early next week.
With the new version we don't need to include the custom k4a.hpp anymore because the 1.2.0 version has the added "get_handle" method for the capture class.

**Describe the solution you'd like**
- Update CMake files with the new version (depth_engine also got a bump to 2.0)
- Remove custom k4a.hpp from this repository and use k4a.hpp from the sdk again

**Describe alternatives you've considered**
none

**Additional context**
none

skalldri, 2019-08-23:
There shouldn't be any work required to support the 1.2.0 SDK, besides removing the .hpp file (and even that isn't truly _required_). The new CMake files will automatically select the most recent compatible version of the SDK that's installed on the machine. The compatibility logic should pick 1.2.0 as a compatible SDK, and the depthengine version check was removed to allow new depthengine versions to be discovered automatically.

Once the .deb package is published next week, I'll consider bumping the required version to 1.2.0 and removing the .hpp file.

RoseFlunder, 2019-08-23:
@skalldri
Don't we have to change this for example?
[https://github.com/microsoft/Azure_Kinect_ROS_Driver/blob/melodic/include/azure_kinect_ros_driver/k4a_ros_device.h#L29](https://github.com/microsoft/Azure_Kinect_ROS_Driver/blob/melodic/include/azure_kinect_ros_driver/k4a_ros_device.h#L29)

From:
`#include "azure_kinect_ros_driver/k4a.hpp"`
to
`#include <k4a/k4a.hpp>`

closed 2019-09-10 by skalldri