microsoft/Azure_Kinect_ROS_Driver #63: Object Boundary "Leak" in RGB-D Image and PointCloud

msr-peng, 2019-08-24:
The problem still exists in the last SDK.
Not already resolved

**Describe the bug**
There is severe overlaps at the boundary of the object in `/rgb_to_depth/image_raw` topic. You can just see my hand below. It just looks like there is another hand close behind my real hand.
[image]
Besides, the pointclouds topic also seems suffer from this boundary problem. It make the object's looks "larger" due to the overlap of the object's boundary. What's more, the pointcloud also has approximate 1 second lag, make real-time application based on Azure Kinect ROS Driver difficult.

**To Reproduce**
My `driver.launch` parameters configurations:
[image]
run
```
roslaunch Azure_Kinect_ROS_Driver/launch/driver.launch
```
and then show `/rgb_to_depth/image_raw` in `rqt_image_view`

**Expected behavior**
A registered RGB-D image with corrrect and sharp boundary. A real-time pointcloud2 data within low lag (just consider Kinect V1 camera's real-time pointcloud data performance in Rviz).

RoseFlunder, 2019-08-24:
I already reported the performance issue (lag of colorized point cloud) in #56. So that problem is confirmed and being looked into.

RoseFlunder, 2019-08-25:
I can confirm that I see the overlaps as well in the colorized depth image and the colorized pointcloud.