microsoft/Azure_Kinect_ROS_Driver #70: "process has died" when roslaunch driver.launch

msr-peng, 2019-09-10:
**Describe the bug & To Reproduce**
After `catkin_make` the `Azure_Kinect_ROS_Driver` ROS package, when run
```
roslaunch driver.launch
```
following errors will occur
[image]

**Expected behavior**
The driver.launch should run successfully.

**Desktop (please complete the following information):**
 - OS: Ubuntu 16.04 (running on Intel NUC mini PC)
 - Version: ROS Kinetic

**Additional context**
I already sucessfully run Azure ROS Driver on my laptop (Ubuntu 16.04 & ROS Kinetic). However, when I do the same thing on my robot mini PC (which is Intel NUC, in Ubuntu 16.04 & ROS Kinetic environment), this error occured. 

I can make sure that the robot mini PC fulfill the hardware requirment to run Azure Kinect. Actually, I followed the Azure Kinect SDK build steps, and can successfully run `k4aviewer` on my mini PC. Here is the screenshot:
[image]

Another thing that might help the troubleshooting is that, when I run `driver.launch` on robot mini PC without plugging Azure Kinect camera, the exact same error will occur. So it seems that the error might occur before the sensor is being found.

Could anyone give me any clues to debug it? Any suggestion will also help me. I know Azure Kinect is not officially supported to run in Ubuntu 16.04 & ROS Kinect environment. But currently many robot can only work with this Ubuntu version.

RoseFlunder, 2019-09-10:
I may have a guess.
I had similar problems in my fork when using the latest sensor sdk 1.2.0 & body tracking sdk 0.9.2 (I added body tracking data publishing in my fork).
This repository still includes a modified "k4a.hpp" based on the 1.1.0 version of the sensor sdk which adds a method to get a captures handle.
This method is now also available in the 1.2.0 release of the sensor sdk and so a modified version is no longer needed.
After I deleted this modified header file and instead included the "k4a.hpp" from the sdk, everything worked again.
I have a branch in my repository with just this change.
As you can see in the pull request, nothing changes except requiring the current sdk version and deleting the custom k4a.hpp in favor of the sdk version:
https://github.com/microsoft/Azure_Kinect_ROS_Driver/pull/68

skalldri, 2019-09-11:
@msr-peng can you try the latest updates to the ROS driver and see if this fixes the issue for you?

msr-peng, 2019-09-11:
> I may have a guess.
> I had similar problems in my fork when using the latest sensor sdk 1.2.0 & body tracking sdk 0.9.2 (I added body tracking data publishing in my fork).
> This repository still includes a modified "k4a.hpp" based on the 1.1.0 version of the sensor sdk which adds a method to get a captures handle.
> This method is now also available in the 1.2.0 release of the sensor sdk and so a modified version is no longer needed.
> After I deleted this modified header file and instead included the "k4a.hpp" from the sdk, everything worked again.
> I have a branch in my repository with just this change.
> As you can see in the pull request, nothing changes except requiring the current sdk version and deleting the custom k4a.hpp in favor of the sdk version:
> #68
> 
> EDIT:
> The pull request just got merged.
> Just try to pull the latest changes and rebuild the ROS packages.

Thanks. I'll try the latest updates of ROS driver.

msr-peng, 2019-09-11:
> @msr-peng can you try the latest updates to the ROS driver and see if this fixes the issue for you?

Hey skalldri, I just tried the laster ROS driver, now `driver.launch` can run successfully!

The new little issue is the loop time warning. The acutally loop time is a little longer than expected max loop time. Is it okey for the rgbd image quality? Is there any way to shorter the loop time, make it below the expected max loop time

closed 2019-09-11 by msr-peng

skalldri, 2019-09-11:
@msr-peng that loop timer warning is letting you know that the main sensor loop isn't hitting framerate, so frames from the camera might be dropped and you might see higher latency in the output images. For some reason, this is reasonably common when using the colorized pointcloud mode: it seems to just be taking a really long time to produce a colorized point cloud.

RoseFlunder, 2019-09-12:
@skalldri
I have measured that the loop that fills the point cloud message using the buffer from the point cloud image is taking longer than the creation of the point cloud image from the sdk.
Do you know a way to speed up this loop?
https://github.com/microsoft/Azure_Kinect_ROS_Driver/blob/melodic/src/k4a_ros_device.cpp#L587

RoseFlunder, 2019-09-12:
@msr-peng
To get a better performance build the package with:
catkin_make -DCMAKE_BUILD_TYPE=Release