microsoft/Azure_Kinect_ROS_Driver #71: Getting an error on catkin_make: configuration file not accepted

amamamahmood, 2019-09-10:
Hello,

I have installed Microsoft Kinect Azure SDK on ubuntu 18.04 When I try to build this repo, I get this error on running "catkin_make"

Kindly help me with it

[image]

skalldri, 2019-09-10:
You need to update to the [1.2.0 Azure Kinect SDK](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/usage.md).

closed 2019-09-10 by skalldri

amamamahmood, 2019-09-10:
I try doing it using: 
sudo apt install libk4a1.2-dev
But I get this error:
[image]

Do I need to get rid of libk4a1.1.1 first? If yes, How?

skalldri, 2019-09-10:
Yes, you need to uninstall the old Azure Kinect SDK first.

`sudo apt purge libk4a1.1-dev libk4a1.1` should remove all traces of the
old SDK.