microsoft/Azure_Kinect_ROS_Driver #75: Publish body index map in ROS

zchenpds, 2019-09-16:
**Is your feature request related to a problem? Please describe.**
No body segmentation result is provided by this driver.

**Describe the solution you'd like**
Publish [body index map](https://docs.microsoft.com/en-us/azure/Kinect-dk/body-index-map) in ROS if `body_tracking_enabled=true`

**Describe alternatives you've considered**
(None)

**Additional context**
[image]

RoseFlunder, 2019-09-24:
@zchenpds 
I just created a pull request which adds a topic for the body index map.
You could test if this meets your requirements.

zchenpds, 2019-09-25:
@RoseFlunder 
Thanks a lot for creating the pull request. I should have made it clearer in the **Additional Context** that I wanted to use the index map to find from the depth image a group of points corresponding to a particular body segment of a particular body. Although an RGB index map is visually pleasing, it's costlier to query from an RGB index map whether a depth image pixel belongs to a particular body. Are there any other reasons that you prefer RGB over a mono image?

RoseFlunder, 2019-09-25:
This was just my initial idea to have the same color the body markers for the joints have.
But I have no problem in changing it up.
I can change it a mono image.
What about the values? Body IDs & 255 as background?
I think body IDs make more sense than just the raw index what do you think?

zchenpds, 2019-09-25:
Sorry for the confusion. Thanks again for your help.

RoseFlunder, 2019-09-25:
No problem. I am glad to help.
You can try the updated pull request again and give feedback if this matches your requirements now.

skalldri, 2019-10-01:
Fixed by #81

closed 2019-10-01 by skalldri