microsoft/Azure_Kinect_ROS_Driver #78: ROS Drivers crash when using export ROS_MASTER_URI and export ROS_HOSTNAME

menonmeg, 2019-09-19:
**Describe the bug**
So I have gotten the Azure Kinect working on Ubuntu 16.04 without any issue aside from the issue of the image processing thread running behind:

[ WARN] [1568899738.323945977]: Image processing thread is running behind.
Expected max loop time: 0.200000000
Actual loop time: 0.214532173

I want to use the Azure Kinect on a mobile robot's ros master, and thus perform the steps outlined in http://wiki.ros.org/ROS/Tutorials/MultipleMachines to export the mobile robot's ros master. However, when I perform the commands outlined in the reproduce section, the terminal immediately crashes without any log file generated that I can find. In my k4atypes.h file, the logging level is set to most severe. 

**To Reproduce**
Steps to reproduce the behavior:
1. Connect Azure Kinect to PC running Ubuntu 16.04
2. Perform following commands to gain access to USB connected Kinect
sudo chmod 666 /dev/bus/usb/001/*
sudo chmod 66 /dev/bus/usb/002/*
3. Ensure you are connected to remote roscore by pinging remote address
3. Source the workspace containing the Azure Kinect ROS Drivers and then run the following commands:
export ROS_MASTER_URI=http://192.168.1.101:11311 
//Insert your own external roscore address here
export ROS_HOSTNAME=192.168.1.198 
//Insert your own PC address here
4. Run roslaunch azure_kinect_ros_driver driver.launch

**Expected behavior**
Launch file does not crash

**Logs**
Unable to find log files

Please also enable *DEBUG* level logging of the ROS node.
See [roscpp/Overview/Logging](http://wiki.ros.org/roscpp/Overview/Logging) for how to enable ROS DEBUG logs.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Ubuntu
 - Version 16.04

**Additional context**
I pulled the changes made in https://github.com/microsoft/Azure_Kinect_ROS_Driver/issues/70 however that did not fix the issue either

RoseFlunder, 2019-09-19:
I use this package myself in a multi machine setup without any problems.
PC #1: Running ROS Core + Connected 2 azure kinects and for each Kinect running an instance of the kinect driver node
PC #2: Connected 1 kinect and running the driver node with ROS Master URI to PC #1

So your problem is related to something else.
How are you starting the kinect ros driver?
Are you using a specific launch file?
Node or Nodelet?
Which parameters?

menonmeg, 2019-09-19:
Hello, thanks for the quick response!
1. Which version of azure kinect sensor sdk installed?
I am using version 1.2 of the azure kinect sensor sdk
2. How are you starting the kinect ros driver?
I am using the "roslaunch azure_kinect_ros_driver driver.launch" command to run the node as found in the launch folder of the azure kinect ros driver package
3. Are you using a specific launch file?
No, I am using the one given in the package, except I have turned off the rgb and rgb point cloud, kept the depth enabled and the point cloud enabled, and I have changed the depth mode to WFOV_2X2BINNED. Changing the depth mode does not fix the issue. 
4. Node or Nodelet?
Yes the driver node from what it looks like in the launch file
5. Which parameters?
"depth_enabled"           default="true"
"depth_mode"              default="WFOV_2X2BINNED"
"color_enabled"           default="false"
"color_resolution"        default="720P"
"point_cloud"             default="true"
"rgb_point_cloud"         default="false"
"required"                default="false" 
"sensor_sn"               default=""
"recording_file"          default=""
"recording_loop_enabled"  default="false"
6. Is your melodic branch of this package up to date?
Yes, I pulled the changes to the package last week after changes were made to deal with the process died (#70) change
7. Other things to note
The mobile robot that I want to use as ros master also runs Ubuntu 16.04 however it does not have the Azure Kinect Sensor SDK installed or the driver node, that only exists on my second PC from which I am launching the driver node. Also I attempted adding these lines to the launch file to deal with the 32FC1 issue from https://github.com/microsoft/Azure_Kinect_ROS_Driver/issues/67 however that did not help anything as well.

RoseFlunder, 2019-09-19:
Okay. Looks fine.
What happens when you only start roscore on the mobile robot without any another nodes and then run the azure kinect ros driver from the other pc?

If it still dies, what happens when you set the ROS_MASTER_URI to http://localhost:11311?
Does it work then?

menonmeg, 2019-09-19:
I am running other nodes but I do not think they subscribe to any of the topics from the Azure Kinect Ros Driver yet. So I tried killing the ros master on my mobile robot and the just running ros core as you suggested and it seems to be not crashing anymore! But I do need to have everything else running, so is there somewhere I should look to figure out what is causing conflict and killing the kinect driver process?

menonmeg, 2019-09-19:
I got it! I added in a namespace of "kinect" to the driver.launch file and that solved the issue. Thanks for the help.

RoseFlunder, 2019-09-19:
Nice that you figured it out.
I have two more tips for you:
1. If you don't want to modify any files so that can pull new changes in more easily you can specifiy a namespace like this:

> ROS_NAMESPACE=kinect roslaunch azure_kinect_ros_driver driver.launch

2. If you want to get a better performance you should build the package with:
catkin_make -DCMAKE_BUILD_TYPE=Release
This really speeds up the image processing thread and the warning about missing the loop rate should appear less often.

closed 2019-09-20 by menonmeg