microsoft/Azure_Kinect_ROS_Driver #97: Issue running two kinect cameras on a same machine through ROS driver

Michae1Park, 2019-11-07:
We are running two Kinects through the ROS driver, and it seems like there is an issue connecting them through the USB. The weird part is that sometimes the cameras connect without an issue, or restarting the cameras multiple times "fixes" the problem. 

We have already made usbfs adjustments following https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/485. 

We are on ROS melodic and Ubuntu 18.04. 

Below is the error message.

[ INFO] [1572984480.118636104]: K4A Parameters:                                                      
[ INFO] [1572984480.119257532]: sensor_sn - std::string : 000032293712                               
[ INFO] [1572984480.119309027]: depth_enabled - bool : 1                                             
[ INFO] [1572984480.119326127]: depth_mode - std::string : NFOV_UNBINNED                             
[ INFO] [1572984480.119345351]: color_enabled - bool : 1                                             
[ INFO] [1572984480.119364867]: color_format - std::string : bgra                                    
[ INFO] [1572984480.119378943]: color_resolution - std::string : 1440P                               
[ INFO] [1572984480.119390089]: fps - int : 30                                                       
[ INFO] [1572984480.119400281]: point_cloud - bool : 0                                               
[ INFO] [1572984480.119410932]: rgb_point_cloud - bool : 0                                           
[ INFO] [1572984480.119421161]: point_cloud_in_depth_frame - bool : 1                                
[ INFO] [1572984480.119431340]: tf_prefix - std::string : camera1_                                   
[ INFO] [1572984480.119441309]: recording_file - std::string :                                       
[ INFO] [1572984480.119451336]: recording_loop_enabled - bool : 0                                    
[ INFO] [1572984480.119461344]: body_tracking_enabled - bool : 0                                     
[ INFO] [1572984480.119484974]: body_tracking_smoothing_factor - float : 0                           
[ INFO] [1572984480.119495871]: imu_rate_target - int : 0                                            
[ INFO] [1572984480.119506017]: rescale_ir_to_mono8 - bool : 0                                       
[ INFO] [1572984480.119517051]: ir_mono8_scaling_factor - float : 1                                  
[ INFO] [1572984480.119527413]: wired_sync_mode - int : 1                                            
[ INFO] [1572984480.124689331]: Found 2 sensors                                                      
[ INFO] [1572984480.124728091]: Searching for sensor with serial number: 000032293712                
[ INFO] [1572984480.147432833]: K4A Parameters:                                                      
[ INFO] [1572984480.147880693]: sensor_sn - std::string : 000269493712                               
[ INFO] [1572984480.147898463]: depth_enabled - bool : 1                                             
[ INFO] [1572984480.147913883]: depth_mode - std::string : NFOV_UNBINNED                             
[ INFO] [1572984480.147930725]: color_enabled - bool : 1                                             
[ INFO] [1572984480.147955945]: color_format - std::string : bgra                                    
[ INFO] [1572984480.147981587]: color_resolution - std::string : 1440P                               
[ INFO] [1572984480.147993070]: fps - int : 30                                                       
[ INFO] [1572984480.148007800]: point_cloud - bool : 0                                               
[ INFO] [1572984480.148062133]: rgb_point_cloud - bool : 0                                           
[ INFO] [1572984480.148076705]: point_cloud_in_depth_frame - bool : 1                                
[ INFO] [1572984480.148090036]: tf_prefix - std::string : camera2_                                   
[ INFO] [1572984480.148103613]: recording_file - std::string :                                       
[ INFO] [1572984480.148117913]: recording_loop_enabled - bool : 0                                    
[ INFO] [1572984480.148131730]: body_tracking_enabled - bool : 0                                     
[ INFO] [1572984480.148163465]: body_tracking_smoothing_factor - float : 0                           
[ INFO] [1572984480.148178002]: imu_rate_target - int : 0                                            
[ INFO] [1572984480.148192588]: rescale_ir_to_mono8 - bool : 0                                       
[ INFO] [1572984480.148207981]: ir_mono8_scaling_factor - float : 1
[ INFO] [15729844
[2019-11-05 15:08:00.156] [error] [t=13559] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/us
bcommand.c (30): TraceLibUsbError(). /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcomman
d.c (447): libusb_claim_interface(usbcmd->libusb, usbcmd->interface) returned LIBUSB_ERROR_BUSY in us
b_cmd_create                                                                                         
[2019-11-05 15:08:00.156] [error] [t=13559] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/us
bcommand.c (30): TraceLibUsbError(). /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcomman
d.c (489): libusb_release_interface(usbcmd->libusb, usbcmd->interface) returned LIBUSB_ERROR_NOT_FOUN
D in usb_cmd_destroy                                                                                 
[2019-11-05 15:08:00.156] [error] [t=13559] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth_mcu/depth_mcu.c (68): usb_cmd_create(USB_DEVICE_DEPTH_PROCESSOR, device_index, NULL, &depthmcu->usb_cmd) returned failure in depthmcu_create()
[2019-11-05 15:08:00.157] [error] [t=13559] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (133): depthmcu_create(index, &device->depthmcu) returned failure in k4a_device_open()
[ERROR] [1572984480.157068584]: Failed to open K4A device at index 0
[ INFO] [1572984480.382604587]: K4A[0] : 000269493712
[ INFO] [1572984480.444531693]: K4A[1] : 000032293712
[2019-11-05 15:08:00.495] [error] [t=13551] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (30): TraceLibUsbError(). /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (447): libusb_claim_interface(usbcmd->libusb, usbcmd->interface) returned LIBUSB_ERROR_BUSY in usb_cmd_create
[2019-11-05 15:08:00.495] [error] [t=13551] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (30): TraceLibUsbError(). /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (489): libusb_release_interface(usbcmd->libusb, usbcmd->interface) returned LIBUSB_ERROR_NOT_FOUND in usb_cmd_destroy
[2019-11-05 15:08:00.495] [error] [t=13551] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth_mcu/depth_mcu.c (68): usb_cmd_create(USB_DEVICE_DEPTH_PROCESSOR, device_index, NULL, &depthmcu->usb_cmd) returned failure in depthmcu_create()
[2019-11-05 15:08:00.495] [error] [t=13551] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (133): depthmcu_create(index, &device->depthmcu) returned failure in k4a_device_open()
[ERROR] [1572984480.495722995]: Failed to open K4A device at index 1
[ERROR] [1572984480.495788941]: Failed to open a K4A device. Cannot continue.
[ INFO] [1572984480.495818889]: Setting wired sync mode: 1
[ INFO] [1572984480.495837643]: Setting RGB Camera Format: bgra
[ INFO] [1572984480.495859076]: Setting RGB Camera Resolution: 1440P
[ INFO] [1572984480.495881217]: Setting Depth Camera Mode: NFOV_UNBINNED
[ INFO] [1572984480.495898688]: Setting Camera FPS: 30
[ INFO] [1572984480.495918625]: Using default IMU rate. Setting to maximum: 1600 Hz.
[ INFO] [1572984480.495946535]: Setting Target IMU rate to 1600 (desired: 1600)
[ INFO] [1572984480.496042274]: K4A Started
[ERROR] [1572984480.500203295]: Failed to open a K4A device. Cannot continue.
[ INFO] [1572984480.500265224]: Setting wired sync mode: 2
[ INFO] [1572984480.500347781]: Setting RGB Camera Format: bgra
[ INFO] [1572984480.500394846]: Setting RGB Camera Resolution: 1440P
[ INFO] [1572984480.500436188]: Setting Depth Camera Mode: NFOV_UNBINNED
[ INFO] [1572984480.500478589]: Setting Camera FPS: 30
[ INFO] [1572984480.500527819]: Using default IMU rate. Setting to maximum: 1600 Hz.
[ INFO] [1572984480.500578731]: Setting Target IMU rate to 1600 (desired: 1600)
[ INFO] [1572984480.500701062]: K4A Started
^C[camera2/node-6] killing on exit

GabrielQuik, 2019-11-08:
### **Having a similar issue myself, I modified my driver.launch file:** 
```
<launch>
  <arg name="tf_prefix"         default="" />               <!-- Prefix added to tf frame IDs. It typically contains a trailing '_' unless empty. -->

  <param name="robot_description"
    command="xacro $(find azure_kinect_ros_driver)/urdf/azure_kinect.urdf.xacro tf_prefix:=$(arg tf_prefix)" />

  <node name="joint_state_publisher1" pkg="joint_state_publisher" type="joint_state_publisher" />
  <node name="robot_state_publisher1" pkg="robot_state_publisher" type="state_publisher" />

  <arg name="depth_enabled"           default="true" />           <!-- Enable or disable the depth camera -->
  <arg name="depth_mode"              default="NFOV_UNBINNED" />  <!-- Set the depth camera mode, which affects FOV, depth range, and camera resolution. See Azure Kinect documentation for full details. Valid options: NFOV_UNBINNED, NFOV_2X2BINNED, WFOV_UNBINNED, WFOV_2X2BINNED, and PASSIVE_IR -->
  <arg name="color_enabled"           default="true" />           <!-- Enable or disable the color camera -->
  <arg name="color_format"            default="bgra" />           <!-- The format of RGB camera. Valid options: bgra, jpeg -->
  <arg name="color_resolution"        default="3072P" />          <!-- Resolution at which to run the color camera. Valid options: 720P, 1080P, 1440P, 1536P, 2160P, 3072P -->
  <arg name="fps"                     default="15" />              <!-- FPS to run both cameras at. Valid options are 5, 15, and 30 -->
  <arg name="point_cloud"             default="true" />           <!-- Generate a point cloud from depth data. Requires depth_enabled -->
  <arg name="rgb_point_cloud"         default="false" />           <!-- Colorize the point cloud using the RBG camera. Requires color_enabled and depth_enabled -->
  <arg name="point_cloud_in_depth_frame" default="false" />        <!-- Whether the RGB pointcloud is rendered in the depth frame (true) or RGB frame (false). Will either match the resolution of the depth camera (true) or the RGB camera (false). -->
  <arg name="required"                default="false" />          <!-- Argument which specified if the entire launch file should terminate if the node dies -->
  <!-- <arg name="sensor_sn"               default="000319193812" />               Sensor serial number. If none provided, the first sensor will be selected -->
  <arg name="recording_file"          default="" />               <!-- Absolute path to a mkv recording file which will be used with the playback api instead of opening a device -->
  <arg name="recording_loop_enabled"  default="false" />          <!-- If set to true the recording file will rewind the beginning once end of file is reached -->
  <arg name="body_tracking_enabled"           default="false" />  <!-- If set to true the joint positions will be published as marker arrays -->
  <arg name="body_tracking_smoothing_factor"  default="0.0" />    <!-- Set between 0 for no smoothing and 1 for full smoothing -->
  <arg name="rescale_ir_to_mono8"  default="false" />    <!-- Whether to rescale the IR image to an 8-bit monochrome image for visualization and further processing. A scaling factor (ir_mono8_scaling_factor) is applied. -->
  <arg name="ir_mono8_scaling_factor"  default="1.0" />    <!-- Scaling factor to apply when converting IR to mono8 (see rescale_ir_to_mono8). If using illumination, use the value 0.5-1. If using passive IR, use 10. -->
  <arg name="imu_rate_target" default="0"/>                       <!-- Desired output rate of IMU messages. Set to 0 (default) for full rate (1.6 kHz). --> 

  <node pkg="azure_kinect_ros_driver" type="node" name="azure_kinect_ros_driver1" output="screen" required="$(arg required)">
    <param name="depth_enabled"     type="bool"   value="$(arg depth_enabled)" />
    <param name="depth_mode"        type="string" value="$(arg depth_mode)" />
    <param name="color_enabled"     type="bool"   value="$(arg color_enabled)" />
    <param name="color_format"      type="string" value="$(arg color_format)" />
    <param name="color_resolution"  type="string" value="$(arg color_resolution)" />
    <param name="fps"               type="int"    value="$(arg fps)" />
    <param name="point_cloud"       type="bool"   value="$(arg point_cloud)" />
    <param name="rgb_point_cloud"   type="bool"   value="$(arg rgb_point_cloud)" />
    <param name="point_cloud_in_depth_frame"   type="bool"   value="$(arg point_cloud_in_depth_frame)" />

    <param name="sensor_sn"         type="string" value="000319193812" /> <!--$(arg sensor_sn) -->

    <param name="tf_prefix"         type="string" value="1" />   <!--$(arg tf_prefix)-->

    <param name="recording_file"          type="string" value="$(arg recording_file)" />
    <param name="recording_loop_enabled"  type="bool"   value="$(arg recording_loop_enabled)" />
    <param name="body_tracking_enabled"           type="bool"   value="$(arg body_tracking_enabled)" />
    <param name="body_tracking_smoothing_factor"  type="double" value="$(arg body_tracking_smoothing_factor)" />
    <param name="rescale_ir_to_mono8" type="bool" value="$(arg rescale_ir_to_mono8)" />
    <param name="ir_mono8_scaling_factor" type="double" value="$(arg ir_mono8_scaling_factor)" />
    <param name="imu_rate_target" type="int" value="$(arg imu_rate_target)"/>
  </node>

  <node pkg="azure_kinect_ros_driver" type="node" name="azure_kinect_ros_driver2" output="screen" required="$(arg required)">
[... cut ...]
[ INFO] [1573209961.900803228]: Setting the time base using a k4a_imu_sample_t sample. Assuming the transmission delay to be 5 ms.
[ WARN] [1573209962.318769084]: Image processing thread is running behind.
Expected max loop time: 0.066666667
Actual loop time: 0.477720524
```
### Then I'm getting this error while the camera is still up and running, I'm getting topics only from one of two cameras.
```[2019-11-08 11:46:03.299] [error] [t=6350] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c (144): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:     49466 type:Depth
```

wirthual, 2019-11-09:
I had similar issues connecting to multiple devices, I am not sure if this is the actual problem but in my code I closed the device when the serial number does not match and then I could use multiple devices.
I am thinking about this part of the code:
```
if (params_.sensor_sn != "")
      {
        if (device.get_serialnum() == params_.sensor_sn)
        {
          k4a_device_ = std::move(device);
          break;
        }
      }
```
If the serial numbers do not match, the device is not closed?
I think it would be useful if the sdk could provide a function to connect to a device by passing in a serial number instead of the index.

RoseFlunder, 2019-11-09:
@wirthual The device should be closed by the destructor when it goes out of scope:
[See device descructor in c++ wrapper](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/include/k4a/k4a.hpp#L1092)

wirthual, 2019-11-09:
I see, thank you for pointing out. I use a variable with a scope that's not only the for loop, probably this was the reason I needed to close it explicit in my code.

Tarekbouamer, 2020-01-20:
I had the same issue in running two KAs at the same time, By default, Linux limits image capturing to a Max_value, in my desktop it was limited to 16 MB. 

To extend the USBFS limit, I manually  moddified the grub 
( **/etc/default/grub**  ) chaning ( **GRUB_CMDLINE_LINUX_DEFAULT="quiet splash"** ) to ==> ( **GRUB_CMDLINE_LINUX_DEFAULT="quiet splash usbcore.usbfs_memory_mb=1000"** ),  

Update the grup ( **sudo update-grub** ) and restart your PC ( **sudo reboot** ).

check if buffer size has successfully changed ( **cat /sys/module/usbcore/parameters/usbfs_memory_mb** )  

All the Best :)

carcormir, 2020-01-23:
@GabrielQuik Did you manage to solve your problem? I am also using two kinects in the same computer and get the same error and after it, both cameras work fine.

> Failed to open K4A device at index 0

I believe it could be the way the two cameras are brought up. Every time the driver is launch, it will try to open all the devices connected to the PC no matter whether they have been already opened or not. This error does not limit the functionality but it would be nice if it did not happen.

RoseFlunder, 2020-01-23:
To prevent this error we would need to remove the option to choose a kinect via serial number because you can only query the serial number of the device if you have opened it via the API.
Already opened devices will produce this error.

carcormir, 2020-01-23:
@RoseFlunder Thanks for your fast response.
Is there a way to launch multiple cameras and avoid this error?

Tarekbouamer, 2020-01-23:
@carcormir can u run two kinects using K4aviewer ? do u get the error .

carcormir, 2020-01-23:
@Tarekbouamer I open two instances of k4aviewer and I am able to run both kinects simultaneously without any errors. The problem comes when trying to launch two cameras using the ROS driver

RoseFlunder, 2020-01-23:
Well the error just means that the device that is opened by the first instances of the driver node can't be opened by the second instance, which is not a problem to worry about.
If you only want the error gone just because of asthehic reasons you could just comment out this line of code in your copy of this package:
https://github.com/microsoft/Azure_Kinect_ROS_Driver/blob/bda2956ddbec3eb1f7fb2b2297c84f11255d1ab6/src/k4a_ros_device.cpp#L163

carcormir, 2020-01-24:
That is what I already did, but I was not sure if there was another way.
Thanks for the help though!
Have a great weekend.

jonra1993, 2020-02-08:
Hi @Michae1Park, did you solve it? I have same problem when I try to open second Kinect camera?

ooeygui, 2020-10-30:
Duplicating to the feature request - https://github.com/microsoft/Azure_Kinect_ROS_Driver/issues/126

working236, 2022-02-23:
> cat /sys/module/usbcore/parameters/usbfs_memory_mb

it seems like that i could not fix the problem with your proposal. i turly get 1000 after your instruction, but i still could open 2 device. Could you give me more advice?

leoneversberg, 2022-10-05:
I seem to have found a workaround solution to launching two kinect camera nodes on the same computer, as mentioned [here](https://github.com/microsoft/Azure_Kinect_ROS_Driver/issues/97#issuecomment-551577760) or in #236 and #210
The solution seems to be a delay between launching the kinect nodes instead of launching them at the same time.
I used [this launch-prefix](https://answers.ros.org/question/233353/set-delay-between-starting-nodes-within-launch-file/?answer=316355#post-id-316355) to create a parameter `node_start_delay` for the `azure_kinect_ros_driver` node which is set to 1 sec for the second camera.

Here is an example driver.launch for two Azure Kinect cameras:

```
<launch>
  <arg name="tf_prefix"         default="" />                       <!-- Prefix added to tf frame IDs. It typically contains a trailing '_' unless empty. -->
  <arg name="overwrite_robot_description" default="true" />         <!-- Flag to publish a standalone azure_description instead of the default robot_descrition parameter-->

  <group if="$(arg overwrite_robot_description)">
    <param name="robot_description"
      command="xacro $(find azure_kinect_ros_driver)/urdf/azure_kinect.urdf.xacro tf_prefix:=$(arg tf_prefix)" />
    <node name="joint_state_publisher" pkg="joint_state_publisher" type="joint_state_publisher" />
    <node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher" />
  </group>

  <group unless="$(arg overwrite_robot_description)">
    <param name="azure_description"
      command="xacro $(find azure_kinect_ros_driver)/urdf/azure_kinect.urdf.xacro tf_prefix:=$(arg tf_prefix)" />
    <node name="joint_state_publisher_azure" pkg="joint_state_publisher" type="joint_state_publisher">
      <remap from="robot_description" to="azure_description" />
    </node>  
    <node name="robot_state_publisher_azure" pkg="robot_state_publisher" type="robot_state_publisher">
      <remap from="robot_description" to="azure_description" />
    </node>
  </group>

  <arg name="depth_enabled"           default="true" />           <!-- Enable or disable the depth camera -->
  <arg name="depth_mode"              default="WFOV_UNBINNED" />  <!-- Set the depth camera mode, which affects FOV, depth range, and camera resolution. See Azure Kinect documentation for full details. Valid options: NFOV_UNBINNED, NFOV_2X2BINNED, WFOV_UNBINNED, WFOV_2X2BINNED, and PASSIVE_IR -->
  <arg name="depth_unit"              default="16UC1" />          <!-- Depth distance units. Options are: "32FC1" (32 bit float metre) or "16UC1" (16 bit integer millimetre) -->
  <arg name="color_enabled"           default="true" />           <!-- Enable or disable the color camera -->
  <arg name="color_format"            default="bgra" />           <!-- The format of RGB camera. Valid options: bgra, jpeg -->
  <arg name="color_resolution"        default="1536P" />          <!-- Resolution at which to run the color camera. Valid options: 720P, 1080P, 1440P, 1536P, 2160P, 3072P -->
  <arg name="fps"                     default="5" />              <!-- FPS to run both cameras at. Valid options are 5, 15, and 30 -->
  <arg name="point_cloud"             default="true" />           <!-- Generate a point cloud from depth data. Requires depth_enabled -->
  <arg name="rgb_point_cloud"         default="true" />           <!-- Colorize the point cloud using the RBG camera. Requires color_enabled and depth_enabled -->
  <arg name="point_cloud_in_depth_frame" default="false" />        <!-- Whether the RGB pointcloud is rendered in the depth frame (true) or RGB frame (false). Will either match the resolution of the depth camera (true) or the RGB camera (false). -->
  <arg name="required"                default="false" />          <!-- Argument which specified if the entire launch file should terminate if the node dies -->
  <arg name="sensor_sn"               default="" />               <!-- Sensor serial number. If none provided, the first sensor will be selected -->
  <arg name="recording_file"          default="" />               <!-- Absolute path to a mkv recording file which will be used with the playback api instead of opening a device -->
  <arg name="recording_loop_enabled"  default="false" />          <!-- If set to true the recording file will rewind the beginning once end of file is reached -->
  <arg name="body_tracking_enabled"           default="false" />  <!-- If set to true the joint positions will be published as marker arrays -->
  <arg name="body_tracking_smoothing_factor"  default="0.0" />    <!-- Set between 0 for no smoothing and 1 for full smoothing -->
  <arg name="rescale_ir_to_mono8"  default="false" />    <!-- Whether to rescale the IR image to an 8-bit monochrome image for visualization and further processing. A scaling factor (ir_mono8_scaling_factor) is applied. -->
  <arg name="ir_mono8_scaling_factor"  default="1.0" />    <!-- Scaling factor to apply when converting IR to mono8 (see rescale_ir_to_mono8). If using illumination, use the value 0.5-1. If using passive IR, use 10. -->
  <arg name="imu_rate_target" default="0"/>                       <!-- Desired output rate of IMU messages. Set to 0 (default) for full rate (1.6 kHz). --> 
  <arg name="wired_sync_mode" default="0"/>                       <!-- Wired sync mode. 0: OFF, 1: MASTER, 2: SUBORDINATE. --> 
[... cut ...]
    <param name="body_tracking_smoothing_factor"  type="double" value="$(arg body_tracking_smoothing_factor)" />
    <param name="rescale_ir_to_mono8" type="bool" value="$(arg rescale_ir_to_mono8)" />
    <param name="ir_mono8_scaling_factor" type="double" value="$(arg ir_mono8_scaling_factor)" />
    <param name="imu_rate_target" type="int" value="$(arg imu_rate_target)"/>
    <param name="wired_sync_mode" type="int" value="$(arg wired_sync_mode)"/>
    <param name="subordinate_delay_off_master_usec" type="int" value="$(arg subordinate_delay_off_master_usec)"/>
  </node>
</launch>
```

jiejie567, 2023-02-02:
Hi, @leoneversberg , I wonder after launching two cameras, did they have the same timestamps for the data.

leoneversberg, 2023-02-21:
> Hi, @leoneversberg , I wonder after launching two cameras, did they have the same timestamps for the data.

I didn't notice any differences in the timestamps or sync issues. I followed the instructions from https://learn.microsoft.com/en-us/azure/kinect-dk/multi-camera-sync and connected both cameras with a 3.5mm cable.