microsoft/Azure_Kinect_ROS_Driver #99: Camera name for topics

jefft0, 2019-11-20:
In the launch file for most cameras, there is a parameter for the camera name which is part of the topic names. I'm especially interested in the point cloud topic. I don't see a parameter for the camera name in the Azure launch file:

https://github.com/microsoft/Azure_Kinect_ROS_Driver/blob/melodic/launch/driver.launch

When I run the camera, `rostopic list` just shows /points2 without a namespace prefix. How do I specify a parameter for the camera name that the Azure will use in publishing topics?

skalldri, 2019-11-20:
I think you can launch the driver in a namespace by setting the `ROS_NAMESPACE` environment variable. For example:

`ROS_NAMESPACE=<camera name> roslaunch azure_kinect_ros_driver driver.launch`

jefft0, 2019-11-21:
I launch the driver from another launch file. How can I modify the launch command to set the namespace?

    <include file="$(find azure_kinect_ros_driver)/launch/driver.launch">
    </include>

skalldri, 2019-11-21:
The [roslaunch syntax guide](http://wiki.ros.org/roslaunch/XML/include) has the answer: use the `ns=""` XML attribute to instruct ROS to launch it in a new namespace. Like this:

```
<include file="$(find azure_kinect_ros_driver)/launch/driver.launch" ns="camera_name" />
```

jefft0, 2019-11-21:
That works! Thanks.

closed 2019-11-21 by jefft0