microsoft/Azure_Kinect_ROS_Driver #100: Topic for 2D depth point cloud

jefft0, 2019-11-22:
The topic /points2 seems to publish the point cloud as a one-dimensional array. I see that the SDK has the function [k4a_transformation_depth_image_to_point_cloud](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/group___functions_ga7385eb4beb9d8892e8a88cf4feb3be70.html#ga7385eb4beb9d8892e8a88cf4feb3be70) for transforming the array. But is there a topic where the Azure driver will directly publish the point cloud as a 2D array?

jefft0, 2019-11-22:
The object passed to the point cloud callback as width 368640 and height 1 . I made a copy of the object and simply set width = 640 and height = 576 so that code will interpret the array as 2D. But is it OK to do that?

closed 2019-12-19 by jefft0

dnlwbr, 2020-12-01:
Hi @jefft0 , how did you solve this issue? I am also wondering how to get a point cloud with the correct height and width.

jefft0, 2020-12-02:
Hi @dnlwbr , we did what I suggested above, to manually set the width and height:
https://github.com/OpenPTrack/open_ptrack_v2/blob/1804/detection/apps/ground_based_people_detector_node.cpp#L130-L134

    if (cloud->height == 1 && cloud->width == azure_width * azure_height) {
      // The Azure provides a 1D array. Simply fix the width and height to be 2D.
      cloud->width = azure_width;
      cloud->height = azure_height;
    }

dnlwbr, 2020-12-02:
Thanks for the quick response. This worked for you? Even if these values are adjusted, data is still one-dimensional, isn't it?

jefft0, 2020-12-02:
Yes, we found that this was good enough to be able to extract the image and use it in some detection libraries, like here:
https://github.com/OpenPTrack/open_ptrack_v2/blob/1804/detection/include/open_ptrack/detection/impl/ground_based_people_detection_app.hpp#L458