microsoft/Azure_Kinect_ROS_Driver #101: 3D mesh model of the azure kinect

bearpaw, 2019-11-22:
It might be a bit irrelevant but is there a 3D mesh model of the Azure Kinect available? Thanks!

ooeygui, 2019-11-22:
Hello @bearpaw ,
Unfortunately, due to our hardware team's policy, we are unable to provide a high resolution mesh for the Azure Kinect. 
The packages includes is a low resolution model which can be used for collision, positioning and 'representational' rendering.

bearpaw, 2019-11-22:
Hi @ooeygui , thank you very much for your response. Yes I think a low resolution model is very useful for robotics application (like you mentioned for collision, positioning and 'representational' rendering). 

Would you please let me know where is this low resolution model located? Thank you!

ooeygui, 2019-11-22:
It is currently constructed using boxes in the URDF. Are you looking for a stand alone model or similar?

bearpaw, 2019-11-22:
Got it, thanks! Is that the best model  publicly available (just to make sure)?

ooeygui, 2019-11-22:
While I cannot confirm the accuracy or suitability of the model, this might meet your needs - https://grabcad.com/library/kinect-4-azure-1.

I'm sorry I can't be more helpful here :-(

christian-rauch, 2021-02-05:
An alternative mesh model without the cover can be found at: https://grabcad.com/library/camera-azure-kinect-dk-1

I also would like to see proper support for an accurate mesh model of the sensor in this repo.

@ooeygui Is it possible to ask the hardware team to verify this model and maybe ask the creator to include it in this repo?

ooeygui, 2021-03-29:
The Azure Kinect team just released a high fidelity model - https://github.com/microsoft/Azure-Kinect-Sensor-SDK/tree/develop/assets. 

This needs to be converted to dae, and added as a mesh resource to the URDF.

christian-rauch, 2021-07-27:
Is there any progress on this? Converting the camera body to DAE should be quite easy. But someone from the engineering team should verify that the frames and transformations on the model are correct.

vstadnytskyi-FDA, 2023-03-06:
> The Azure Kinect team just released a high fidelity model - https://github.com/microsoft/Azure-Kinect-Sensor-SDK/tree/develop/assets.
> 
> This needs to be converted to dae, and added as a mesh resource to the URDF.

I opened it in SolidWorks and I am getting two errors in the drawing
[image]