microsoft/Azure_Kinect_ROS_Driver #102: Point cloud has NaN for X,Y values

jefft0, 2019-11-26:
I subscribe to the point cloud topic "points2". Many of the points have X and Y values that are NaN. With other sensors, I have seen when the Z value has NaN, but the X and Y values always map out a grid. (I launch the Azure driver with rgb_point_cloud=false and depth_mode=NFOV_UNBINNED .

Is it normal for point cloud values have NaN for X and Y values? Is there a way to configure the Azure to publish point clouds with valid X and Y values?

skalldri, 2019-11-27:
It's normal for these PointCloud2 messages to contain NaN values. NaN values indicate that no depth was recorded for that pixel in the image. There is no way to prevent this from happening, because these invalid values are actually coming from the sensor itself.

There are a variety of reasons the sensor might not capture a valid depth value for a pixel. Some of these reasons are documented on the [Azure Kinect Documentation](https://docs.microsoft.com/en-us/azure/Kinect-dk/depth-camera#invalidation). The raw sensor output uses a depth value of 0 to indicate invalid pixels. PointCloud2 messages use NaN values to indicate this same thing.

You might want to check out this [ROS Answer](https://answers.ros.org/question/234455/pointcloud2-and-pointfield/) for more information about dense PointCloud2s. We might be able to switch over to using sparse PointClouds in the driver, which would eliminate the NaNs.

jefft0, 2019-11-27:
Thanks for the links. I'm asking because when I call pcl::IntegralImageNormalEstimation::compute, it gives me the error "Input dataset is not from a projective device". I'm assuming this is because the point cloud is not formatted properly. Is that right?

closed 2019-12-19 by jefft0

Pran-Seven, 2022-10-18:
@jefft0 were you able to get a fix?

jefft0, 2022-10-18:
I cannot help you. If there was a solution, I don't remember. That project ended years ago.

Pran-Seven, 2022-10-18:
No worries, thanks a lot!