microsoft/Azure_Kinect_ROS_Driver #104: Weird layer at the top of image_view for rgb images

rajk97, 2019-12-06:
Hi all, 

Lately, when I'm trying to run image_view to visualize /rgb/image_raw, I'm getting a weird filter at the top that looks like below:

[image]

Any idea why this might be happening? Kindly suggest. Thank you.

ooeygui, 2020-01-20:
This appears to be a rendering artifact, not a camera artifact. Can you attempt to view with the camera directly using the Kinect SDK?

rajk97, 2020-01-20:
Thanks for your reply. I did try it with Kinect SDK and it didn't have a problem with it. Any idea of how we can fix this? Please comment. Thanks.

ooeygui, 2020-01-21:
@rajk97 This appears to be a rendering problem on your system - not a camera or Kinect ROS node problem.

closed 2020-01-21 by ooeygui