microsoft/Azure_Kinect_ROS_Driver #105: Calibration in ROS #959

rajk97, 2019-12-06:
Hi all!

I have been trying to calibrate the Azure Kinect following the tutorial specified at:

[http://wiki.ros.org/camera_calibration/Tutorials/MonocularCalibration](http://wiki.ros.org/camera_calibration/Tutorials/MonocularCalibration)

From what I understand, what it does is that it gets calibration information and then communicates with ROS service /camera/set_camera_info to change the intrinsic parameters of the camera. 

I didn't find the camera/set_camera_info service being published by the camera node and hence when I click save after calibrating (following the tutorial), I get the following error:

```
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/camera_calibration/camera_calibrator.py", line 252, in on_mouse
    if self.do_upload():
  File "/opt/ros/melodic/lib/python2.7/dist-packages/camera_calibration/camera_calibrator.py", line 209, in do_upload
    response = self.set_camera_info_service(info)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 439, in __call__
    return self.call(*args, **kwds)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 499, in call
    service_uri = self._get_service_uri(request)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 467, in _get_service_uri
    raise ServiceException("service [%s] unavailable"%self.resolved_name)
rospy.service.ServiceException: service [/camera/set_camera_info] unavailable

```

Is there a way out of it? Kindly suggest any other way that has worked for you. 

ROS Version: Melodic
Ubuntu 18.04

Ros Services running:

> /azure_kinect_ros_driver/get_loggers
/azure_kinect_ros_driver/set_logger_level
/depth/image_raw/compressed/set_parameters
/depth/image_raw/compressedDepth/set_parameters
/depth/image_raw/theora/set_parameters
/depth_to_rgb/image_raw/compressed/set_parameters
/depth_to_rgb/image_raw/compressedDepth/set_parameters
/depth_to_rgb/image_raw/theora/set_parameters
/ir/image_raw/compressed/set_parameters
/ir/image_raw/compressedDepth/set_parameters
/ir/image_raw/theora/set_parameters
/joint_state_publisher/get_loggers
/joint_state_publisher/set_logger_level
/rgb/image_raw/compressed/set_parameters
/rgb/image_raw/compressedDepth/set_parameters
/rgb/image_raw/theora/set_parameters
/rgb_to_depth/image_raw/compressed/set_parameters
/rgb_to_depth/image_raw/compressedDepth/set_parameters
/rgb_to_depth/image_raw/theora/set_parameters
/robot_state_publisher/get_loggers
/robot_state_publisher/set_logger_level
/rosout/get_loggers
/rosout/set_logger_level
/rostopic_20269_1575590186848/get_loggers
/rostopic_20269_1575590186848/set_logger_level

Ros Nodes running:

> /azure_kinect_ros_driver
/joint_state_publisher
/robot_state_publisher
/rosout
/rostopic_20269_1575590186848
 
Appreciate your time and help. Thanks!

ooeygui, 2020-01-20:
The ROS node does not currently support setting the camera info from on site calibration. It does use the factory calibration. It would be interesting to see how different on-site calibration is from the factory calibration. We would love a PR for this.

liangxiaowei00, 2020-12-28:
hi @ooeygui ,the factory calibration always is good enough for using rgb and depth image ?

ooeygui, 2020-12-29:
@liangxiaowei00 I think the answer is a definitive "it depends". The Factory Calibration is good - but it is a compromise. We have seen customer reports of better results by calibrating your specific camera in the specific runtime environment.

christian-rauch, 2021-07-09:
See https://github.com/microsoft/Azure_Kinect_ROS_Driver/pull/200 for calibrating the camera via ROS.

PR #200 (2021-08-10): camera intrinsics calibration

## Fixes #105, #199

### Description of the changes:
- add the camera_info_manager to store/load camera parameters
- add rgbd_launch for image rectification
- ~~all of the changes in https://github.com/microsoft/Azure_Kinect_ROS_Driver/pull/180~~

### Required before submitting a Pull Request:
- [x] I reviewed [CONTRIBUTING.md](https://github.com/microsoft/Azure_Kinect_ROS_Driver/blob/master/CONTRIBUTING.md)
- [x] I [built my changes](https://github.com/microsoft/Azure_Kinect_ROS_Driver/blob/master/docs/building.md) locally
- [x] I tested my changes with a device [... cut]
closed 2021-08-10 by amelhassan