microsoft/Azure_Kinect_ROS_Driver #106: Depth values sent by ROS is significantly degraded.Please help me.

dicaleon, 2019-12-15:
Now I am working on a project using Azure Kinect.
Our project is using ROS to send Azure Kinect’s raw color and raw depth image.
However I found the raw depth image’s resolution is significantly degraded.
It has only ten depth values in the raw depth image get from subscriber.
If I don’t use ROS, it has a lot of depth values in the depth image.

My source code is shown as below.
ーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーー
ros::Subscriber sub = nh.subscribe("/front_center_kinect/rgb/image_raw", 1, &Detector::image_cb, &dec);
ros::Subscriber subdepth = nh.subscribe("/front_center_kinect/depth_to_rgb/image_raw", 1, &Detector::image_depth_cb, &dec);

sensor_msgs::Image::ConstPtr last_depth, last_color;
    void image_depth_cb(const sensor_msgs::Image::ConstPtr& img)
    {
        if (img->header.stamp >= m_depth_timestamp) 
        {
            {	
		    std::lock_guard<std::mutex> lock_depth(m_mtx_depth);
		    m_depth_img = matFromImage(*img);
		    last_depth = img;
		    m_depth_timestamp =img->header.stamp;
            std::ostringstream oss;
            oss << m_depth_timestamp;
            cv::imwrite("img_D\\pic_" + oss.str() +".png", m_depth_img); 
	        }
        }
    }

    void image_cb(const sensor_msgs::Image::ConstPtr& img)
    {
        if (img->header.stamp >= m_color_timestamp) 
        {
	        {
		    std::lock_guard<std::mutex> lock_color(m_mtx_color);
		    m_color_timestamp = img->header.stamp;
		    last_color = img;
		    m_color_timestamp_ns =img->header.stamp.nsec;
		    m_color_timestamp_ns /=10000000;	
            std::cout << "m_color_timestamp_ns=" << m_color_timestamp_ns << std::endl;
            m_color_timestamp_s =img->header.stamp.sec;
		    cv::Mat rgba = matFromImage(*img);
		    cv::cvtColor(rgba, m_current_frame, cv::COLOR_RGBA2BGR);
    	    }
        }
    }
-------------------------------------------------------------------------------------------------------
As you can see, I used opencv to save the Depth image as a png file [cv::imwrite("img_D\\pic_" + oss.str() +".png", m_depth_img); ]. And after that I also used opencv to check the png file's depth values. The values for the png file has only 10 values. I dont know where is the problem. 
The below is my analyse.
1, I should set some parameters in Azure Kinect ROS driver . 
   If so, Please tell me what kind of parameters should be set.
2, Depth image's save method is bad. A good method should be instead of [cv::imwrite("img_D\\pic_" + oss.str() +".png", m_depth_img);].
   If so, Please tell me the  command.
By the way, as for realsense D435, camera_tf_frame is set as "camera_depth_optical_frame".
Does Azure Kinect’s camera_tf_frame need to be changed to “depth_camera_link” or something else?

Thank you very much.

bnjff, 2019-12-18:
Your problem is likely that the depth is a float image (in meters) and opencv truncates it when you save it as PNG. Either rescale to int (e.g. from m to mm) or save it as an array without using opencv.

dicaleon, 2019-12-19:
Oh, I got it. You are right. Ihave solved the problem by rescaling the image to int.
Thank you very much!

closed 2019-12-19 by dicaleon

PannagaS, 2024-02-18:
> Oh, I got it. You are right. Ihave solved the problem by rescaling the image to int. Thank you very much!

Hi, how did you solve this? Can you please paste a code snippet for reference? 
I have saved png files from /rgb/image_rect_color and /depth_to_rgb/hw_registered/image_rect ROS topics. I'm able to see the rgb image properly but the depth is completely blank. It is all 0,0,0. How do I fix this? and how do I rescale this?