microsoft/Azure_Kinect_ROS_Driver #107: image_proc: No output for /mycamera/depth/image_rect

jefft0, 2019-12-16:
The [usage docs](https://github.com/microsoft/Azure_Kinect_ROS_Driver/blob/melodic/docs/usage.md#topics) say to use `image_proc`. My Azure driver is under the namespace mycamera, so I enter:

    ROS_NAMESPACE=mycamera/depth rosrun image_proc image_proc

In RViz, I see the image for the input topic /mycamera/depth/image_raw. I see the topic listed for /mycamera/depth/image_rect, but the image is empty. The terminal shows the error:

    Raw image data from topic '/mycamera/depth/image_raw' has unsupported depth: 32

What does "unsupported depth: 32" mean? Will this prevent making the output image?

ooeygui, 2019-12-16:
Can you tell us more about your environment? Windows/Linux? 

That error comes from the rendering of the image in rViz - the image is a file format that it cannot visualize.

jefft0, 2019-12-16:
The environment is Ubuntu 18.04. I'm using `image_proc` for the depth image because the [usage docs](https://github.com/microsoft/Azure_Kinect_ROS_Driver/blob/melodic/docs/usage.md#topics) tell me to. Should it be possible to view the image in rViz?

bnjff, 2019-12-18:
Use `image_proc/rectify` nodelet; it works. See example in `launch/slam_rtabmap.launch`

ooeygui, 2020-02-29:
Per @bnjff's note

closed 2020-02-29 by ooeygui