microsoft/Azure_Kinect_ROS_Driver #109: Body tracking latency

pengy8, 2020-01-17:
**Is your feature request related to a problem? Please describe.**
The body tracking data in the ROS driver is very useful to my application, but somehow there's about 1-second latency in the reported data. However, the data seen in the body tracking SDK is faster (about 0.25 second). 

**Describe the solution you'd like**
Is there any way to improve the body tracking data latency in the ROS driver to be close to the SDK's speed?

**Describe alternatives you've considered**
NA

**Additional context**
NA

ooeygui, 2020-06-28:
It may be due to the double hop though pub/sub.

RoseFlunder, 2020-06-29:
There is also a huge discussion when it comes to the latency about the body tracking sdk in the sdk repository:
https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/514

I am on mobile at the moment but I remember that one person wrote that there is like 1 buffered frame in the body tracker after creation. He got less latency when he pulled that from the body tracker instance before queimg new frames. Maybe you can test this.

ooeygui, 2020-06-29:
@RoseFlunder Thank you for linking this in.

PR #161 (2021-10-30): Use a separate thread to pop k4abt_tracker result. Fix #109.

## Fixes #109 

### Description of the changes:
- Create a separate thread dedicated to popping the result from `k4abt_tracker_` and publishing body_tracking_data following the suggestion from [this post](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1130#issuecomment-598866234):
> the tracker will process faster if you keep queuing the frame without waiting for previous frame to be finished. Because then the pipeline can process two frames at different stages at the same time.

### Before submitting a Pull Request:
-  [... cut]
commit 39127118 (2021-10-30): Merge pull request #161 from zchenpds/body_tracking_delay

Use a separate thread to pop k4abt_tracker result. Fix #109.
commit 4b0fe11c (2021-10-30): Use a separate thread to pop k4abt_tracker result. Fix #109.
PR #162 (2021-10-30): Use a separate thread to pop k4abt_tracker result. Fix #109.

## Fixes #109 

### Description of the changes:
- Create a separate thread dedicated to popping the result from `k4abt_tracker_` and publishing body_tracking_data following the suggestion from [this post](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1130#issuecomment-598866234):
> the tracker will process faster if you keep queuing the frame without waiting for previous frame to be finished. Because then the pipeline can process two frames at different stages at the same time.

### Required before submitting a Pull Re [... cut]
closed 2021-10-30 by amelhassan