microsoft/Azure_Kinect_ROS_Driver #110: Wrong output from /depth_to_rgb and /rgb_to_depth topics

carcormir, 2020-01-20:
Hi,

My problem is simple, after running the driver.launch file, I launch rviz to visualize the camera output. When I plot the output the color image does not match the depth image ( /depth_to_rgb or /rgb_to_depth topics).

I thought this could be a calibration issue between the ir and color cameras but I have tried other two kinect azure platforms and the same issue occurs. 

This is a screenshot of my rviz session:

[image]

These are the arguments of my launch file:

`  <arg name="depth_enabled"           default="true" />           
  <arg name="depth_mode"              default="NFOV_UNBINNED" />  
  <arg name="color_enabled"           default="true" />           
  <arg name="color_format"            default="bgra" />           
  <arg name="color_resolution"        default="1080P" />          
  <arg name="fps"                     default="15" />              
  <arg name="point_cloud"             default="true" />           
  <arg name="rgb_point_cloud"         default="true" />           
  <arg name="point_cloud_in_depth_frame" default="true" />        
  <arg name="required"                default="false" />          
  <arg name="sensor_sn"               default="" />               
  <arg name="recording_file"          default="" />               
  <arg name="recording_loop_enabled"  default="false" />          
  <arg name="body_tracking_enabled"           default="false" />  
  <arg name="body_tracking_smoothing_factor"  default="0.0" />    
  <arg name="rescale_ir_to_mono8"  default="true" />    
  <arg name="ir_mono8_scaling_factor"  default="0.65" />    
  <arg name="imu_rate_target" default="0"/>                       
`

Any idea on how to approach to this? Should I do my own extrinsic calibration?

Thanks in advance for the help

AswinkarthikeyenAK, 2020-09-03:
Hi @ooeygui ,
Is there a way to synchronize the color and the depth images in ros? Also, How do you find the devices depth settings? For instance I would like to know if the depth setting is in [m] or [mm]?

Thanks

DanielTakeshi, 2022-03-18:
Also curious about how to synchronize color and depth images? The images have different sizes.
Also depth should generally be in meters (but this can be verified by measuring with a ruler).