microsoft/Azure_Kinect_ROS_Driver #111: Help need: body tracking data is not being published on one machine

Thompson104, 2020-01-24:
Hello All,

We have two computers both running Ubuntu 18.04 and ROS Melodic. However, one of the computers cannot publish the body tracking data (no topic is shown in rostopic list).

1) We have the Azure SDK and body tracking SDK installed successfully;

I wonder if anyone could help us in figuring out what the problems might be?

Thank you,

ooeygui, 2020-01-24:
The Body Tracking SDK requires a certain level of GPU performance. Can you identify hardware differences?

ooeygui, 2020-01-24:
The Body Tracking SDK requires minimum GPU performance. Are they the same machine category?

Thompson104, 2020-01-25:
No they are not. The computer that can publish the body tracking data is a very high-end computer (about $9000 in total). The one that doesn't publish is about $1000. 

So if the hardware doesn't satisfy the minimum running requirement, the data is not published? Please correct me if I'm wrong. If that is the case, I wonder what would be the minimum requirement of the GPU to publish the body tracking data?

Thank you a lot for your help. I really spent a lot of time on this.

ooeygui, 2020-01-25:
Here are the [hardware specs required for the Kinect](https://docs.microsoft.com/en-us/azure/Kinect-dk/system-requirements). The Body Tracking SDK requires a later generation Nvidia GPU and a dedicated *USB3 Port*.

Thompson104, 2020-01-27:
Thank you for the clarification. The lower-end PC has a USB 3.0 port. I'm kind of curious, if the PC doesn't meet the minimum requirement, why the topic is not published at all?

Thanks.

RoseFlunder, 2020-01-27:
Hi,
which versions of the respective SDKs did you install?
I just updated the CMakeLists.txt to the current version numbers (1.3.0 for sensor sdk and 1.0.0 for body tracking sdk) in this branch in my forked repo:
https://github.com/RoseFlunder/Azure_Kinect_ROS_Driver/tree/feat/bump-sdk-versions-1.3-1.0

Maybe you could make sure that these versions are installed via "apt list --installed | grep k4a" in a terminal and try out my linked branch.

Thompson104, 2020-01-27:
Thank you all. Updating the lib to the most current version seems to solve the problem. FYI, the graphics board we have is NVIDIA Corporation GP107GL [Quadro P1000] and I can confirm the data can be published via the topic body_tracking_data

closed 2020-01-27 by Thompson104