microsoft/Azure_Kinect_ROS_Driver #113: questions regarding the mapping between joint index and marker id (body_tracking_data)

Thompson104, 2020-01-28:
Hello all,

I'm trying to get different joint positions in the camera frame based on the body tracking data. I wonder what would be the mapping between the marker ID and joint name? **For example, if I have 120 as the marker ID, what would be the joint name?** (I know the marker ID is 20 based on the discussion [here](https://github.com/microsoft/Azure_Kinect_ROS_Driver/issues/87)  )

I'm reading from the official joint index document as listed [`here`](https://docs.microsoft.com/en-us/azure/kinect-dk/body-joints). It looks like we have joint index from 0 to 32, so I assume there should be 33 marker IDs for one person. However, based on the discussion from issue 87 and the ros topic data, it looks like we only have 25 IDs published for each person. So why is it that we only have 25 marker IDs?

Thank you for your help.

RoseFlunder, 2020-01-28:
When using Body tracking SDK 1.0.0 there should be 32 markers for each person.
In example ID 100 to 131 for the person with body id 1 . 200 to 231 for the person with body id 2 etc.

Joint id = Marker id % 100
Example: marker id 120 means body id 1 and joint id 20. According to the docs this is the "Ankle left" joint.

If you have less joints you may have installed an older version of the body tracking sdk because some joints are additions in newer versions.

Thompson104, 2020-01-28:
Thank you. I have updated my body tracking lib and now I have 32 markers. So is it safe to assume for a given marker ID, the corresponding joint can be found based on this mapping [(`here`](https://docs.microsoft.com/en-us/azure/kinect-dk/body-joints)?

Appreciate your quick response.

RoseFlunder, 2020-01-28:
Yes thats correct.
We use the body id and joint id from the sdk to calculate the marker id and you can always reverse the calculation to get the original body id and joint id from the sdk as explained.

closed 2020-01-29 by Thompson104