microsoft/Azure_Kinect_ROS_Driver #114: Question: how to get depth value from pixel location of RGB image?

Thompson104, 2020-01-29:
Hi All,

Basically I'm confused about how the data is stored in the the following three topics:  /depth/image_raw, /rgb/image_raw and /rgb_to_depth/image_raw

1)  /depth/image_raw has the data type sensor_msgs/Images [here](http://docs.ros.org/melodic/api/sensor_msgs/html/msg/Image.html).  Based on the definition of sensor_msgs/images, I couldn't figure out where can I store the depth information. For example, for a given pixel location (x,y) of the depth image, where can I find the depth value? The uint8[] data says it stores  # actual matrix data, size is (step * rows), is this related to the depth value?

2) For the topic: /rgb_to_depth/image_raw, I assume this would be the topic for me to find the depth value from a given pixel location of a RGB image (I could be wrong, my major is mehcanical:( ). However, I'm not sure where the depth information is stored among all the members of the sensor_msgs/Images. I wonder if you may kindly help me?

Thank you.

skalldri, 2020-01-29:
1. In a `sensor_msgs/Image`, the `data` field is a byte array containing data for all the pixels in an image. The thing is, images come in many varieties. Maybe it's just grayscale, so it only has one channel per pixel. Maybe it's full color, so it has three channels per pixel? Or maybe it's not a color image at all, it's a depth map (like `/depth/image_raw`). And when you know how many channels are in the image, what data type are they? Unsigned integers? Floats? How many bits of precision? Without knowing all of this, you can't lookup pixels in the data buffer.

    So ROS encodes extra information about each image in the message to help work this out. The `encoding` string tells you what format to expect in the `data` field. In the case of `/depth/image_raw`, the `encoding` should be `32FC1`, which means it a 32-bit float (`32F`) with a single channel (`C1`) . So, you should be able to cast the `data` array to an array of `float` and expect it to contain (`width` * `height`) values (which is one float per pixel). The float values are the depth in meters.

    `step` tells you how many bytes to expect in a single row. `step` is helpful because pixels can have more than one channel, and one channel might not be a single byte, so to quickly lookup the starting pixel for a particular row you can just multiply `step` by the desired row index to get the location of the first pixel for that row in `data`.

2. `/rgb_to_depth` transforms the RGB camera into the Depth camera's perspective. The images on those topics will all be RGB images and will not contain depth data. The pixels in the `/rgb_to_depth` topics will correspond with pixels in the `/depth` topics. So if you look at pixel [100, 100] in `/rgb_to_depth` then the corresponding depth for that pixel can be found in `/depth` at [100, 100].

    `/depth_to_rgb` does the same thing for the Depth camera: it transforms the depth image into the RGB camera's perspective. So if you look at pixel [100, 100] in `/depth_to_rgb`, the corresponding color for that pixel can be found at pixel [100, 100] in `/rgb`.

    It should be noted that this process (called reprojection) introduces artifacts into the depth or RGB image, depending on which one is being reprojected.

Thompson104, 2020-01-29:
Thank you so much for this detailed explanation. I should buy you lunch man lol. Really appreciate it.

closed 2020-01-29 by Thompson104