microsoft/Azure_Kinect_ROS_Driver #115: Questions: two GPUs installed, how does the Kinect know which GPU to use?

Thompson104, 2020-02-05:
Hello,

My PC just installed another GPU for the Azure Kinect. I wonder if I need to change some parameters of this ROS package to tell the SDK which GPU it should use?

Currently if I run nvidia-smi I have the following results:
`+-----------------------------------------------------------------------------+
| NVIDIA-SMI 440.33.01    Driver Version: 440.33.01    CUDA Version: 10.2     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  Quadro P1000        On   | 00000000:03:00.0  On |                  N/A |
| 34%   50C    P0    N/A /  N/A |    382MiB /  4031MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
|   1  GeForce RTX 208...  On   | 00000000:04:00.0 Off |                  N/A |
|  0%   40C    P8    18W / 250W |      1MiB / 11019MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                       GPU Memory |
|  GPU       PID   Type   Process name                             Usage      |
|=============================================================================|
|    0      1457      G   /usr/lib/xorg/Xorg                            65MiB |
|    0      1599      G   /usr/bin/gnome-shell                          52MiB |
|    0      1872      G   /usr/lib/xorg/Xorg                           164MiB |
|    0      2017      G   /usr/bin/gnome-shell                          95MiB |
`

Thank you.

RoseFlunder, 2020-02-05:
As far as I know you can only specify which GPU will be used for the Body Tracking SDK.
To do this you will need to change the code in this line because the node itself doesn't provide a parameter for this yet:
https://github.com/microsoft/Azure_Kinect_ROS_Driver/blob/melodic/src/k4a_ros_device.cpp#L305

The[create()](https://microsoft.github.io/Azure-Kinect-Body-Tracking/release/1.x.x/classk4abt_1_1tracker_ad453000dc74bf9412b78e48f03bec433.html#ad453000dc74bf9412b78e48f03bec433) method takes an optional [configuration](https://microsoft.github.io/Azure-Kinect-Body-Tracking/release/1.x.x/structk4abt__tracker__configuration__t.html) that you can provide.
In this configuration you can specify the [GPU device ID](https://microsoft.github.io/Azure-Kinect-Body-Tracking/release/1.x.x/structk4abt__tracker__configuration__t_a663351b13a0caa2b0240abcbcbcd7b14.html#a663351b13a0caa2b0240abcbcbcd7b14) which is just the integer you see in your nvidia smi output in front of the GPUs name.

ooeygui, 2020-02-06:
Thanks @RoseFlunder for answering on this. I would appreciate a pull request on this topic - I'm unable to test multi-gpu scenarios at the moment.

Thompson104, 2020-02-06:
I just tested this code with the following code as suggested by RoseFlunder:
```
k4abt_tracker_configuration_t k4abt_config = K4ABT_TRACKER_CONFIG_DEFAULT;
k4abt_config.gpu_device_id = 0;
k4abt_tracker_ = k4abt::tracker::create(calibration_data_.k4a_calibration_,k4abt_config);
k4abt_tracker_.set_temporal_smoothing(params_.body_tracking_smoothing_factor);
```
I specifically put 0(default) for testing based on the discussion [here](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/260). However, I run into the following errors:

`[2020-02-06 11:38:02.591] [error] [t=8928] [K4ABT] /home/vsts/work/1/s/src/TrackerHost/TrackerHost.cpp (132): Create(). The current depth mode is not supported by the K4ABT SDK!
[2020-02-06 11:38:02.591] [error] [t=8928] [K4ABT] /home/vsts/work/1/s/src/sdk/k4abt.cpp (38): tracker->Create(sensor_calibration, config) returned failure in k4abt_tracker_create()
terminate called after throwing an instance of 'k4a::error'
  what():  Failed to create k4abt tracker!
`

I'm curious why the selection of GPU would cause an error. I used the driver.launch with the depth enabled and I have tried all the four depth_mode (NFOV_ and WFOV_) and they all return the same error as listed above. 

When I run `apt list --installed | grep k4a,` I have the following results:
`k4a-tools/bionic,now 1.3.0 amd64 [installed]
libk4a1.3/bionic,now 1.3.0 amd64 [installed,automatic]
libk4a1.3-dev/bionic,now 1.3.0 amd64 [installed]
libk4abt1.0/bionic,now 1.0.0 amd64 [installed,automatic]
libk4abt1.0-dev/bionic,now 1.0.0 amd64 [installed]
`

ooeygui, 2020-02-06:
I'm asking the Azure Kinect SDK team to take a look at this. 
(I'm on the Microsoft Edge Robotics team who maintains the ROS connector; but not the SDK itself)

In the mean time, I suspect that you need to change the ROS parameters for the depth in order for Body Tracking to work.

RoseFlunder, 2020-02-06:
Did you change it exactly at the line I linked?
Just asking because maybe you pasted the code before the `calibration_data_.k4a_calibration_` is correctly initialized.

I have only a single GPU and can confirm that is working when using device id 0 (my only valid device id).
```
k4abt_tracker_configuration_t k4abt_config = K4ABT_TRACKER_CONFIG_DEFAULT;
k4abt_config.gpu_device_id = 0;
k4abt_tracker_ = k4abt::tracker::create(calibration_data_.k4a_calibration_,k4abt_config);
k4abt_tracker_.set_temporal_smoothing(params_.body_tracking_smoothing_factor);
```
If I set a wrong device id like 1, I will get an expected Cuda error:
CUDA failure 10: invalid device ordinal

Thompson104, 2020-02-06:
I believe I have put them into the correct position. However, I could be wrong (my major is mechanical). I have uploaded the k4a_ros_device.cpp file [here](https://drive.google.com/open?id=1EpYOWo36_Mpl7z4KzKxstcEloxQb66Gd)

Starting from line 301, I have the following:
```
#if defined(K4A_BODY_TRACKING)
  // When calibration is initialized the body tracker can be created with the device calibration
  if (params_.body_tracking_enabled)
  {
  	k4abt_tracker_configuration_t k4abt_config = K4ABT_TRACKER_CONFIG_DEFAULT;
  	k4abt_config.gpu_device_id = 0;
    k4abt_tracker_ = k4abt::tracker::create(calibration_data_.k4a_calibration_,k4abt_config);
    k4abt_tracker_.set_temporal_smoothing(params_.body_tracking_smoothing_factor);
  }
#endif
```

RoseFlunder, 2020-02-06:
Looks good to me.
Sorry, I can't reproduce the error.
Does the error still happen if comment out this line?
`k4abt_config.gpu_device_id = 0;`
Because then it would just be the default config which is also used if don't pass a config to the create method of the tracker.

RoseFlunder, 2020-02-06:
Hmm I can't really test anything because I only have a single GPU, but from the looks of it, it seems like GPU device ID 0 in the SDK may use the GPU the nvidia-smi output lists as 1 and SDK GPU device ID 0 may be the one nvidia-smi lists as 0.
This is my guess because if you set it to 0, nvidia-smi prints out that GPU 1 is also used from the node.