microsoft/Azure_Kinect_ROS_Driver #117: Launch camera false when run command ssh on different PC

tuan97anton, 2020-02-28:
**- I have 2 PC Linux ( Ubuntu 18, ROS Melodic ). PC 1 ( I58200, ram 16GB ) connects directly with camera Azure kinect , PC 2 ( my laptop ) connect to PC 1 through SSH . 
- I run cmt : roslaunch azure_kinect_ros_driver driver.launch** 
**- On PC1 :** 
`... logging to /home/sti/.ros/log/72725fb4-592b-11ea-8ea1-04ea56e5db11/roslaunch-AGV-14325.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.4:34299/

SUMMARY
========

PARAMETERS
 * /azure_kinect_ros_driver/body_tracking_enabled: False
 * /azure_kinect_ros_driver/body_tracking_smoothing_factor: 0.0
 * /azure_kinect_ros_driver/color_enabled: True
 * /azure_kinect_ros_driver/color_format: bgra
 * /azure_kinect_ros_driver/color_resolution: 720P
 * /azure_kinect_ros_driver/depth_enabled: True
 * /azure_kinect_ros_driver/depth_mode: WFOV_UNBINNED
 * /azure_kinect_ros_driver/fps: 5
 * /azure_kinect_ros_driver/imu_rate_target: 0
 * /azure_kinect_ros_driver/ir_mono8_scaling_factor: 1.0
 * /azure_kinect_ros_driver/point_cloud: True
 * /azure_kinect_ros_driver/point_cloud_in_depth_frame: False
 * /azure_kinect_ros_driver/recording_file: 
 * /azure_kinect_ros_driver/recording_loop_enabled: False
 * /azure_kinect_ros_driver/rescale_ir_to_mono8: False
 * /azure_kinect_ros_driver/rgb_point_cloud: True
 * /azure_kinect_ros_driver/sensor_sn: 
 * /azure_kinect_ros_driver/subordinate_delay_off_master_usec: 0
 * /azure_kinect_ros_driver/tf_prefix: 
 * /azure_kinect_ros_driver/wired_sync_mode: 0
 * /robot_description: <?xml version="1....
 * /rosdistro: melodic
 * /rosversion: 1.14.3

NODES
  /
    azure_kinect_ros_driver (azure_kinect_ros_driver/node)
    joint_state_publisher (joint_state_publisher/joint_state_publisher)
    robot_state_publisher (robot_state_publisher/state_publisher)

auto-starting new master
process[master]: started with pid [14338]
ROS_MASTER_URI=http://192.168.1.4:11311

setting /run_id to 72725fb4-592b-11ea-8ea1-04ea56e5db11
process[rosout-1]: started with pid [14349]
started core service [/rosout]
process[joint_state_publisher-2]: started with pid [14356]
process[robot_state_publisher-3]: started with pid [14357]
process[azure_kinect_ros_driver-4]: started with pid [14358]
[ WARN] [1582785373.630906633]: The 'state_publisher' executable is deprecated. Please use 'robot_state_publisher' instead
[ INFO] [1582785373.714381596]: K4A Parameters:
[ INFO] [1582785373.714919571]: sensor_sn - std::string : 
[ INFO] [1582785373.714963715]: depth_enabled - bool : 1
[ INFO] [1582785373.714982461]: depth_mode - std::string : WFOV_UNBINNED
[ INFO] [1582785373.715013803]: color_enabled - bool : 1
[ INFO] [1582785373.715046843]: color_format - std::string : bgra
[ INFO] [1582785373.715062646]: color_resolution - std::string : 720P
[ INFO] [1582785373.715109958]: fps - int : 5
[ INFO] [1582785373.715126957]: point_cloud - bool : 1
[ INFO] [1582785373.715144418]: rgb_point_cloud - bool : 1
[ INFO] [1582785373.715159458]: point_cloud_in_depth_frame - bool : 0
[ INFO] [1582785373.715173636]: tf_prefix - std::string : 
[ INFO] [1582785373.715187926]: recording_file - std::string : 
[ INFO] [1582785373.715202251]: recording_loop_enabled - bool : 0
[ INFO] [1582785373.715216804]: body_tracking_enabled - bool : 0
[... 287 lines cut ...]
[ INFO] [1582785508.197611124]: 			 Translation: -46.9588, -28.7056, 3.47741
[ INFO] [1582785508.197630135]: 			 Rotation[0]: 0.0022701, 0.0201766, -0.999794
[ INFO] [1582785508.197651059]: 			 Rotation[1]: -0.999997, 0.000346836, -0.00226356
[ INFO] [1582785508.197670980]: 			 Rotation[2]: 0.000301093, 0.999796, 0.0201774
[ INFO] [1582785508.197690006]: 	 IMU (IMU to Color):
[ INFO] [1582785508.197705242]: 		 Extrinsics:
[ INFO] [1582785508.197726276]: 			 Translation: -28.6, -2.51928, -47.0843
[ INFO] [1582785508.197745173]: 			 Rotation[0]: 0.0022701, -0.999997, 0.000301093
[ INFO] [1582785508.197764460]: 			 Rotation[1]: 0.0201766, 0.000346836, 0.999796
[ INFO] [1582785508.197786015]: 			 Rotation[2]: -0.999794, -0.00226356, 0.0201774
[ INFO] [1582785508.197856434]: STARTING CAMERAS
**[2020-02-27 13:38:28.478] [error] [t=14589] ../src/dewrapper/dewrapper.c (154): depth_engine_start_helper(). Depth engine create and initialize failed with error code: 204.
[2020-02-27 13:38:28.478] [error] [t=14589] ../src/dewrapper/dewrapper.c (160): deresult == K4A_DEPTH_ENGINE_RESULT_SUCCEEDED returned failure in depth_engine_start_helper()
[2020-02-27 13:38:28.478] [error] [t=14589] ../src/dewrapper/dewrapper.c (194): depth_engine_start_helper(dewrapper, dewrapper->fps, dewrapper->depth_mode, &depth_engine_max_compute_time_ms, &depth_engine_output_buffer_size) returned failure in depth_engine_thread()
[2020-02-27 13:38:28.478] [error] [t=14552] ../src/dewrapper/dewrapper.c (552): dewrapper_start(). Depth Engine thread failed to start
[2020-02-27 13:38:28.478] [error] [t=14552] ../src/depth/depth.c (398): dewrapper_start(depth->dewrapper, config, depth->calibration_memory, depth->calibration_memory_size) returned failure in depth_start()
[2020-02-27 13:38:28.479] [error] [t=14552] ../src/depth_mcu/depth_mcu.c (359): cmd_status == CMD_STATUS_PASS returned failure in depthmcu_depth_stop_streaming()
[2020-02-27 13:38:28.479] [error] [t=14552] ../src/depth_mcu/depth_mcu.c (362): depthmcu_depth_stop_streaming(). ERROR: cmd_status=0x00000063
[2020-02-27 13:38:28.479] [error] [t=14552] ../src/sdk/k4a.c (895): depth_start(device->depth, config) returned failure in k4a_device_start_cameras()
terminate called after throwing an instance of 'k4a::error'
  what():  Failed to start cameras!
[azure_kinect_ros_driver-4] process has died [pid 14552, exit code -6, cmd /home/sti/catkin_ws/devel/lib/azure_kinect_ros_driver/node __name:=azure_kinect_ros_driver __log:=/home/sti/.ros/log/c229d42e-592b-11ea-8ea1-04ea56e5db11/azure_kinect_ros_driver-4.log].
log file: /home/sti/.ros/log/c229d42e-592b-11ea-8ea1-04ea56e5db11/azure_kinect_ros_driver-4*.log`**

**I hope gives for your support . Thank you very much !!**

RoseFlunder, 2020-02-28:
Try setting the "DISPLAY" environment variable after connecting via ssh.
Before running roslauch execute:
export DISPLAY=:0

tuan97anton, 2020-03-02:
> Try setting the "DISPLAY" environment variable after connecting via ssh.
> Before running roslauch execute:
> export DISPLAY=:0

I tried and It work. Thank you very much ! :))

tuan97anton, 2020-03-11:
I am sorry . I can't start camera Azure kinect thourgh SSH . Although , I added export DISPLAY=:0 in ~/.bashrc . My camera connect with PC on Robot , and it don't have display .

RoseFlunder, 2020-03-11:
As far as I know the bashrc environment variables will not be set when connecting via ssh.
Try to set the display variable explicitly after connecting via ssh with the export command.

tuan97anton, 2020-03-12:
I tried : export DISPLAY=:0 , and I check : echo $DISPLAY ( result :0 ) . But I run : `roslaunch azure_kinect_ros_driver driver.launch` . It don't work , the error code is still as I said above .

RoseFlunder, 2020-03-12:
Could you check the value of the display variable when the screen is connected?

tuan97anton, 2020-03-18:
yes. Icheck :  echo $DISPLAY  -->result :0

tuan97anton, 2020-03-23:
@RoseFlunder 
Thank your support ! I found my issues . After I turn on `automatic login` on ubuntu and `export DISPLAY=:0` , I can start camera vis SSH .

closed 2020-03-23 by tuan97anton

RoseFlunder, 2020-03-23:
Thanks for your feedback.
That makes sense. I was trying to reproduce it but I already had automatic login enabled so I didn't ran into this issue.
The X-Server is needed for open gl and will be started at login I think.

abc619539082, 2022-05-14:
Hi . I have met a similar problem.
I tried to roslaunch on a PC (ubuntu18.04 ros-melodic but no gui)without a screen through SSH. And I met the same bug.
I turn on automatic login but it doesn't make sense.
Maybe it is impossible for driver.launch to run on a PC without a screen or gui?
Thank you!