microsoft/Azure_Kinect_ROS_Driver #121: [melodic][Kinect v4a][rtabmap]Did not receive data since 5 seconds!

colorDogs, 2020-03-23:
Device: kinect v4 (v4a)
Command:  roslaunch azure_kinect_ros_driver slam_rtabmap.launch 

launch file : [launch](https://raw.githubusercontent.com/microsoft/Azure_Kinect_ROS_Driver/melodic/launch/slam_rtabmap.launch)

the launch file about rtabmap:
` 
  <include file="$(find rtabmap_ros)/launch/rtabmap.launch" ns="rtabmap" >
    <arg name="rgb_topic"          value="/k4a/rgb_to_depth/image_rect" />
    <arg name="depth_topic"        value="/k4a/depth/image_rect" />
    <arg name="camera_info_topic"  value="/k4a/depth/camera_info" />
    <arg name="approx_sync"        value="false" />
    <arg name="frame_id"           value="camera_base" />
    <arg name="args"               value="--delete_db_on_start --Odom/ResetCountdown 2" />
  </include>`

source code:
- [kinect v4  ros wrapper](https://github.com/microsoft/Azure_Kinect_ROS_Driver/blob/melodic/launch/slam_rtabmap.launch)     branch[melodic]
- rtabmap      branch[melodic-devel]
- rtabmap_ros     branch[melodic-devel]

 roslaunch azure_kinect_ros_driver slam_rtabmap.launch 
... logging to /home/lamer/.ros/log/7f2a2fbc-6d07-11ea-acfa-50eb7167c20d/roslaunch-lamer-Mi-Gaming-Laptop-15-6-6633.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://lamer-Mi-Gaming-Laptop-15-6:40323/

SUMMARY
========

PARAMETERS
 * /k4a/k4a_ros_bridge/color_enabled: True
 * /k4a/k4a_ros_bridge/color_resolution: 1536P
 * /k4a/k4a_ros_bridge/depth_enabled: True
 * /k4a/k4a_ros_bridge/depth_mode: NFOV_UNBINNED
 * /k4a/k4a_ros_bridge/fps: 30
 * /k4a/k4a_ros_bridge/point_cloud: False
 * /k4a/k4a_ros_bridge/required: True
 * /k4a/k4a_ros_bridge/rgb_point_cloud: False
 * /k4a/manager/num_worker_threads: 16
 * /robot_description: <?xml version="1....
 * /rosdistro: melodic
 * /rosversion: 1.14.4
 * /rtabmap/rtabmap/rgbd_odometry/approx_sync: False
 * /rtabmap/rtabmap/rgbd_odometry/config_path: 
 * /rtabmap/rtabmap/rgbd_odometry/frame_id: camera_base
 * /rtabmap/rtabmap/rgbd_odometry/ground_truth_base_frame_id: 
 * /rtabmap/rtabmap/rgbd_odometry/ground_truth_frame_id: 
 * /rtabmap/rtabmap/rgbd_odometry/guess_frame_id: 
 * /rtabmap/rtabmap/rgbd_odometry/guess_min_rotation: 0.0
 * /rtabmap/rtabmap/rgbd_odometry/guess_min_translation: 0.0
 * /rtabmap/rtabmap/rgbd_odometry/odom_frame_id: odom
 * /rtabmap/rtabmap/rgbd_odometry/publish_tf: True
 * /rtabmap/rtabmap/rgbd_odometry/queue_size: 10
 * /rtabmap/rtabmap/rgbd_odometry/subscribe_rgbd: False
 * /rtabmap/rtabmap/rgbd_odometry/wait_for_transform_duration: 0.2
 * /rtabmap/rtabmap/rgbd_odometry/wait_imu_to_init: False
 * /rtabmap/rtabmap/rtabmap/Mem/IncrementalMemory: true
 * /rtabmap/rtabmap/rtabmap/Mem/InitWMWithAllNodes: false
 * /rtabmap/rtabmap/rtabmap/approx_sync: False
 * /rtabmap/rtabmap/rtabmap/config_path: 
 * /rtabmap/rtabmap/rtabmap/database_path: ~/.ros/rtabmap.db
 * /rtabmap/rtabmap/rtabmap/frame_id: camera_base
 * /rtabmap/rtabmap/rtabmap/ground_truth_base_frame_id: 
 * /rtabmap/rtabmap/rtabmap/ground_truth_frame_id: 
 * /rtabmap/rtabmap/rtabmap/landmark_angular_variance: 9999.0
 * /rtabmap/rtabmap/rtabmap/landmark_linear_variance: 0.0001
 * /rtabmap/rtabmap/rtabmap/map_frame_id: map
[... 313 lines cut ...]
/rtabmap/rtabmap/rtabmap subscribed to (exact sync):
   /rtabmap/rtabmap/odom,
   /k4a/rgb_to_depth/image_rect,
   /k4a/depth/image_rect,
   /k4a/depth/camera_info,
   /rtabmap/rtabmap/odom_info
[ WARN] [1584968968.219677953]: /rtabmap/rtabmap/rtabmapviz: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). Parameter "approx_sync" is false, which means that input topics should have all the exact timestamp for the callback to be called.
/rtabmap/rtabmap/rtabmapviz subscribed to (exact sync):
   /rtabmap/rtabmap/odom,
   /k4a/rgb_to_depth/image_rect,
   /k4a/depth/image_rect,
   /k4a/depth/camera_info,
   /rtabmap/rtabmap/odom_info
[ WARN] [1584968972.343166219]: /rtabmap/rtabmap/rgbd_odometry: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. Parameter "approx_sync" is false, which means that input topics should have all the exact timestamp for the callback to be called.
/rtabmap/rtabmap/rgbd_odometry subscribed to (exact sync):
   /k4a/rgb_to_depth/image_rect,
   /k4a/depth/image_rect,
   /k4a/depth/camera_info
[ WARN] [1584968972.661113566]: /rtabmap/rtabmap/rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). Parameter "approx_sync" is false, which means that input topics should have all the exact timestamp for the callback to be called.
/rtabmap/rtabmap/rtabmap subscribed to (exact sync):
   /rtabmap/rtabmap/odom,
   /k4a/rgb_to_depth/image_rect,
   /k4a/depth/image_rect,
   /k4a/depth/camera_info,
   /rtabmap/rtabmap/odom_info

ch2000pro, 2020-03-25:
same problem , how is it going?

ooeygui, 2020-04-09:
Looking at this during our April sprint.

colorDogs, 2020-04-11:
> Looking at this during our April sprint.

Thanks,
waiting for your good news

Doolly, 2020-06-09:
still having same issues, any good news?

ooeygui, 2020-06-10:
This work item got bumped from our April sprint due to changes in our developer conference and related announcements.

helenol, 2020-06-11:
Investigating on a slightly higher-end machine. :) Lenovo T490s, Ubuntu 18.04.

Works "out of the box" for me. All start-up output the same except what's noted below:
Notes on potential causes:
* `melodic` branch of this repo. 
* `libk4a1.3`, *not* 1.4. Maybe this is a difference?
* The `ros-melodic-rtabmap` package, *not* the latest from the repo.
* I have slightly newer firmware on the akdk than you.
Mine:
```
[ INFO] [1591869956.382742008]: RGB Version: 1.6.107
[ INFO] [1591869956.382776217]: Depth Version: 1.6.78
[ INFO] [1591869956.382790298]: Audio Version: 1.6.14
[ INFO] [1591869956.382807981]: Depth Sensor Version: 6109.7.0
```
Yours:
```
[ INFO] [1584968956.831521478]: RGB Version: 1.6.102
[ INFO] [1584968956.831584452]: Depth Version: 1.6.75
[ INFO] [1584968956.831621226]: Audio Version: 1.6.14
[ INFO] [1584968956.831642600]: Depth Sensor Version: 6109.7.0
```

But @ooeygui was able to repro. 

So my guess is it's one of the following, in order of decreasing likelihood:
* libk4a version
* rtabmap version (packaged on apt vs melodic-devel from source)
* sensor firmware version
* computer slowness differences

I'll update libk4a to 1.4 and see if it breaks.

helenol, 2020-06-11:
libk4a1.4 made no difference.

But I was able to succesfully repro.
Steps to repro:
add line
```xml
     <param name="imu_rate_target"   value="100" />
```
to line 49 in the slam_rtabmap.launch file.

This then gives me the same error you got:
`
[ WARN] [1591871608.967814553]: /rtabmap/rtabmap/rgbd_odometry: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. Parameter "approx_sync" is false, which means that input topics should have all the exact timestamp for the callback to be called.
/rtabmap/rtabmap/rgbd_odometry subscribed to (exact sync):
   /k4a/rgb_to_depth/image_rect,
   /k4a/depth/image_rect,
   /k4a/depth/camera_info
`

Which gives a hint of what the problem (*and the solution*) is. If you have too few IMU messages, OR you're dropping messages (which is I think what's happening for you, 1600 Hz is way above what ROS can handle), the IMU messages are no longer at the exact same timestamp as the images. So it ignores the IMU messages.

Fix:
Change line 60 in the slam_rtabmap.launch file to be true:
```xml
    <arg name="approx_sync"        value="true" />
```

Nothing wrong with the driver, just a misconfiguration of the launch file for rtabmap.

Making a PR. Going to change rate to 100 Hz anyway in this launch file because no one needs an IMU at 1600 Hz. rtabmap is running at like 1 hz at any rate.

ooeygui, 2020-06-11:
@helenol in my local build I migrated to 1.4 and updated the firmware. I'll check that in after we migrate the CI builds to the new CI system. It didn't have any effect on the scenario.

I do believe this is related to systems on the low end of the Azure Kinect requirements.

Thanks for your IMU config change.

ooeygui, 2020-06-12:
We have committed fixes which reduce the overhead of the Kinect. However, the Kinect requires a relatively high end system, so you may still encounter this problem.

closed 2020-06-12 by ooeygui

hariharan382, 2022-07-19:
You said the reason that it was due to time stamp in frames ,but how did you solve it

ooeygui, 2022-07-19:
@hariharan382 The solution was to reduce the data overhead so it worked on a wider range of systems. If you are still encountering this, there are numerous reasons; including CPU overhead, USB Bus bandwidth, even driver versions.

hariharan382, 2022-07-20:
How can I reduce it , I am getting Odom at 30hz rgbd at 5hz, scan at 5hz then which parameters I should adjust to correct overhead