microsoft/Azure_Kinect_ROS_Driver #122: Can I use K4Arecord in ROS driver to generate a .mkv file?

SladeLu, 2020-03-30:
I need to synchronize azure with my other sensors. So I am wondering if any implements to act like k4arecord in AZURE_KINECT_SDK but controling by ROS nodes to start recording and create a .mkv file with the same timpstamp in ROS.

ooeygui, 2020-03-30:
I'm sorry, I don't follow your request.

zchenpds, 2020-03-31:
I am also interested in this feature. I guess what we want is the capability of playing back a .mkv file in synchronization with a bag file. 

Maybe we could add a parameter `recording_enabled` to allow the user of this driver to optionally record (as a .mkv file) the captures and/or IMU samples plus, e.g., their associated ROS timestamps (or some time offset). The recorded ROS timestamps will enable the .mkv file to be played back in synchronization with the simulated ROS time that is being published, say, by `rosbag play <bag-files> --clock`.

Unfortunately, this driver is based on SDK version 1.3 for now while the [C++ API for recording](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/release/1.4.x/include/k4arecord/record.hpp) is not available until 1.4. I believe it is the Body Tracking SDK 1.0.0 which depends on sensor SDK 1.3 that is preventing us from upgrading to sensor SDK version 1.4. When do you anticipate that the sensor SDK for this driver to be upgraded to 1.4? @ooeygui  Thank you!