microsoft/Azure_Kinect_ROS_Driver #125: Random crash happens after topic subscription

carcormir, 2020-04-07:
Hi,
I am able to initialize the camera driver without a problem. However, as soon as I have a node subscribing to one of the topics, the driver crashes. Sometimes it happens and sometimes is does not.
Also, if I have to sensors launched with independent drivers and in their own namespace, sometimes one of them or even both may crash when a node subscribes to one of the topics.
For example:
Cam1 provides topics /cam1/rgb/...
Cam2 provides topics /cam2/rgb/...

If a node subscribes to /cam1/rgb..., Cam1 AND Cam2 may crash.

This my launch file for the driver:

`

<launch>
  <arg name="tf_prefix"         default="1_" />               

  <arg name="sensor"                      default="camera" />
  <arg name="rgb_rect"                    default="1" />
  <arg name="depth_rect"                  default="1" />
  <arg name="depth_to_rgb_rect"           default="1" />
  <arg name="rgb_to_depth_rect"           default="1" />
  <arg name="ir_rect"                     default="1" />
  <arg name="point_cloud"                 default="true" />                       
  <arg name="rgb_point_cloud"             default="true" />                   

  <arg name="depth_enabled"                       default="true" />           
  <arg name="depth_mode"                          default="WFOV_2X2BINNED" />  
  <arg name="color_enabled"                       default="true" />           
  <arg name="color_format"                        default="bgra" />           
  <arg name="color_resolution"                    default="1440P" />          
  <arg name="fps"                                 default="15" />             
  <arg name="sensor_sn"                           default="5235532" />     
  <arg name="point_cloud_in_depth_frame"          default="true" />        
  <arg name="required"                            default="false" />          
  <arg name="recording_file"                      default="" />               
  <arg name="recording_loop_enabled"              default="false" />          
  <arg name="body_tracking_enabled"               default="false" />  
  <arg name="body_tracking_smoothing_factor"      default="0.0" />    
  <arg name="rescale_ir_to_mono8"                 default="true" />              
  <arg name="ir_mono8_scaling_factor"             default="0.5" />           
  <arg name="imu_rate_target"                     default="0"/>                       
  <arg name="wired_sync_mode"                     default="0"/>                       
  <arg name="subordinate_delay_off_master_usec"   default="0"/>     

  
  <group ns="$(arg sensor)" >
    
    <node pkg="nodelet" type="nodelet" name="manager" args="manager" output="screen">
      <param name="num_worker_threads" value="16" />
    </node>

    
    <node if="$(arg rgb_rect)"
          pkg="nodelet" type="nodelet" name="rectify_rgb"
          args="load image_proc/rectify manager --no-bond"
          respawn="true">
      <remap from="image_mono"  to="rgb/image_raw" />
      <remap from="image_rect"  to="rgb/image_rect_color" />
    </node>

    
    <node if="$(arg depth_rect)"
          pkg="nodelet" type="nodelet" name="rectify_depth"
          args="load image_proc/rectify manager --no-bond"
          respawn="true">
      <remap from="image_mono"  to="depth/image_raw" />
      <remap from="image_rect"  to="depth/image_rect" />

      <param name="interpolation" value="0" />
    </node>

    
    <node if="$(arg rgb_to_depth_rect)"
          pkg="nodelet" type="nodelet" name="rectify_rgb_to_depth"
          args="load image_proc/rectify manager --no-bond"
          respawn="true">
      <remap from="image_mono"  to="rgb_to_depth/image_raw" />
      <remap from="image_rect"  to="rgb_to_depth/image_rect" />

      
    </node>

    
    <node if="$(arg depth_to_rgb_rect)"
          pkg="nodelet" type="nodelet" name="rectify_depth_to_rgb"
          args="load image_proc/rectify manager --no-bond"
          respawn="true">
      <remap from="image_mono"  to="depth_to_rgb/image_raw" />
      <remap from="image_rect"  to="depth_to_rgb/image_rect" />

      
    </node>

    
    <node if="$(arg ir_rect)"
          pkg="nodelet" type="nodelet" name="rectify_ir"
          args="load image_proc/rectify manager --no-bond"
          respawn="true">
      <remap from="image_mono"  to="ir/image_raw" />
      <remap from="image_rect"  to="ir/image_rect" />
    </node>

    <group if="$(arg point_cloud)">
      
      <node unless="$(arg rgb_point_cloud)"
            pkg="nodelet" type="nodelet" name="point_cloud_xyz"
            args="load depth_image_proc/point_cloud_xyz manager --no-bond"
            respawn="true">
        <remap from="image_rect"  to="depth/image_rect" />
      </node>

      <group if="$(arg rgb_point_cloud)">
        
        <node pkg="nodelet" type="nodelet" name="depth_register"
              args="load depth_image_proc/register manager --no-bond"
              respawn="true">
        </node>

        
        <node pkg="nodelet" type="nodelet" name="point_cloud_xyzrgb"
              args="load depth_image_proc/point_cloud_xyzrgb manager --no-bond"
              respawn="true">
        </node>
      </group>
    </group>

    <node pkg="nodelet" type="nodelet" name="k4a_ros_bridge"
          args="load Azure_Kinect_ROS_Driver/K4AROSBridgeNodelet manager --no-bond"
          respawn="true">
      <param name="depth_enabled"                     type="bool"   value="$(arg depth_enabled)" />
      <param name="depth_mode"                        type="string" value="$(arg depth_mode)" />
      <param name="color_enabled"                     type="bool"   value="$(arg color_enabled)" />
      <param name="color_format"                      type="string" value="$(arg color_format)" />
      <param name="color_resolution"                  type="string" value="$(arg color_resolution)" />
      <param name="fps"                               type="int"    value="$(arg fps)" />
      <param name="point_cloud"                       type="bool"   value="$(arg point_cloud)" />
      <param name="rgb_point_cloud"                   type="bool"   value="$(arg rgb_point_cloud)" />
      <param name="point_cloud_in_depth_frame"        type="bool"   value="$(arg point_cloud_in_depth_frame)" />
      <param name="sensor_sn"                         type="string" value="$(arg sensor_sn)" />
      <param name="tf_prefix"                         type="string" value="$(arg tf_prefix)" />
      <param name="recording_file"                    type="string" value="$(arg recording_file)" />
      <param name="recording_loop_enabled"            type="bool"   value="$(arg recording_loop_enabled)" />
      <param name="body_tracking_enabled"             type="bool"   value="$(arg body_tracking_enabled)" />
      <param name="body_tracking_smoothing_factor"    type="double" value="$(arg body_tracking_smoothing_factor)" />
      <param name="rescale_ir_to_mono8"               type="bool"   value="$(arg rescale_ir_to_mono8)" />
      <param name="ir_mono8_scaling_factor"           type="double" value="$(arg ir_mono8_scaling_factor)" />
      <param name="imu_rate_target"                   type="int"    value="$(arg imu_rate_target)"/>
      <param name="wired_sync_mode"                   type="int"    value="$(arg wired_sync_mode)"/>
      <param name="subordinate_delay_off_master_usec" type="double" value="$(arg subordinate_delay_off_master_usec)"/>

    </node>

  </group>

</launch>
`
The subscription is done in a node that does not have anything else but the subscription and a dummy callback.

This is the error that comes up:
`[2020-04-07 13:15:50.398] [error] [t=107589] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/color/uvc_camerareader.cpp (1300): DecodeMJPEGtoBGRA32(). MJPEG decode failed: -1
[2020-04-07 13:15:50.402] [error] [t=107590] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/queue/queue.c (112): queue_pop(). Queue "Queue_capture" was popped in a disabled state.
[2020-04-07 13:15:50.402] [error] [t=107590] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (269): capturesync_get_capture(device->capturesync, capture_handle, timeout_in_ms) returned failure in k4a_device_get_capture()
terminate called after throwing an instance of 'k4a::error'
  what():  Failed to get capture from device!`

I am working with:
Ubuntu 18.04
Azure kinect SDK 1.3
ROS melodic

Any help would be more than welcome, thanks in advance.

ooeygui, 2020-04-07:
This error is coming from the Kinect SDK. Are you able to run multiple cameras with the SDK directly?

carcormir, 2020-04-07:
I can open two instances of k4aviewer and open two cameras, yes.
I can also launch two cameras with the ROS driver but as I mentioned the problem comes when subscribing to any of the topics like rgb/image_raw or rgb/image_rect_color, among others.

carcormir, 2020-04-08:
Some extra information:
When replicating the same problem on a different computer (a laptop that has two controllers), the issue only appears when after launching both cameras with independent launch file, if one of them is stop the other camera crashes (sometimes) with the same error as mentioned before.

I was wondering if there could be an issue with the way the camera buffer is handled when having multiple cameras. Do they share the buffer? I am referring to line 1288, argument in_buf of [uvc_camerareader.cpp](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/master/src/color/uvc_camerareader.cpp)

Any inputs @ooeygui? 

Thanks in advance!

carcormir, 2020-04-15:
After trying multiple things, I manged to make it work when reducing the resolution from 1440P to 720P when launching the drivers. This removes the error and also a warning that might be the cause of the issue.
The warning message is the following:
`Initializing or re-initializing the device to realtime offset: 1586942326144821792 ns`

Even though I solved my problem it was only temporary fix since after bringing up other parts of my system, the same warning and error happened:

`[ WARN] [1586942399.828546544]: Initializing or re-initializing the device to realtime offset: 1586942326144821792 ns

[2020-04-07 13:15:50.398] [error] [t=107589] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/color/uvc_camerareader.cpp (1300): DecodeMJPEGtoBGRA32(). MJPEG decode failed: -1 [2020-04-07 13:15:50.402] [error] [t=107590] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/queue/queue.c (112): queue_pop(). Queue "Queue_capture" was popped in a disabled state. [2020-04-07 13:15:50.402] [error] [t=107590] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (269): capturesync_get_capture(device->capturesync, capture_handle, timeout_in_ms) returned failure in k4a_device_get_capture() terminate called after throwing an instance of 'k4a::error' what(): Failed to get capture from device!`

I would love to have any type of input. @ooeygui, do you suggest me to move to the SDK repo and ask there or what do you think is best? I am really stuck after a week of trying multiple things and I would really appreciate some help.

Thanks in advance,
Carlos

ooeygui, 2020-04-15:
Is this for a commercial or industrial project? Can you share details?
My team won't be able to dig into this until May. 
We offer an escalation path through http://aka.ms/ros/support, if this is critical - it activates another team.

carcormir, 2020-04-16:
Hi @ooeygui,
Thanks for your reply.
In fact this is part of a commercial/industrial project. But we could wait until May. It is not a critical issue for now but it will be in a near future.
**Description**:
We have one PC, 6 Kinect Azure and two robot arms performing different tasks.
The information from the Kinects go through different ML algorithms (TensorFlow models) and the output, combined with the depth map, is used to generate a pose for the robot/s.

Previously we were using a ROS network with multiple PCs and each of them had a Kinect v2 connected. Now we are moving to one station and the new Kinect Azure.

The specifications of the current PC are:
**Hardware**:
AMD® Ryzen threadripper 2970wx 24-core processor × 48
GeForce RTX 2080 Ti/PCIe/SSE2
64Gb Ram
**Software**:
Ubuntu 18.04.04
ROS melodic
k4a: 1.3
Note: the grub was modified to increase the usb limit (modified the line GRUB_CMDLINE_LINUX_DEFAULT)

**Test 1**: Bring up 2 Kinect Azure on one PC and create multiple nodes subscribing to the topics provided by the ROS driver (node and nodelet version have been tried).
**Results from the test**:
- I am able to do multiple instances of k4aviewer and visualize the output of different cameras at the same time. No complains on this part.
- I am able to launch multiple cameras through the ROS driver with rgb resolution=1440P and fps=30. However, when creating multiple nodes (around 4 nodes per camera) that subscribe to the rectified rgb topic and the rectified depth_to_rgb, the ROS driver crashes with this error:

` [2020-04-07 13:15:50.398] [error] [t=107589] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/color/uvc_camerareader.cpp (1300): DecodeMJPEGtoBGRA32(). MJPEG decode failed: -1 [2020-04-07 13:15:50.402] [error] [t=107590] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/queue/queue.c (112): queue_pop(). Queue "Queue_capture" was popped in a disabled state. [2020-04-07 13:15:50.402] [error] [t=107590] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (269): capturesync_get_capture(device->capturesync, capture_handle, timeout_in_ms) returned failure in k4a_device_get_capture() terminate called after throwing an instance of 'k4a::error' what(): Failed to get capture from device!`

**Test 2**: Bring up 3 Kinect Azure on one PC.
**Results from the test**:
- Same as for Test 1.

- It has been noticed that when reducing the rgb resolution to 720P and the fps to 5, the above error does not appear for test 1 and test 2.

We would really appreciate any help on how to move on to being able to achieve higher resolution and fps.

Thanks in advance for the help. 
Please, let us know if further information is needed.

ooeygui, 2020-04-16:
I'm going to see if I can find someone from the Azure Kinect SDK team; although they do not have experience with the ROS codebase.

carcormir, 2020-04-17:
Hi @ooeygui, thanks for your response. We are looking forward to hearing from you.
In the meantime, we keep trying different things to figure out the root of the issue. I will keep you posted if we make any progress.

Just another note to my previous description. Before the error rises, I constantly get this message:
`[ WARN] [1586942399.828546544]: Initializing or re-initializing the device to realtime offset`
I do not know if that can be an indicative of something but it is a repeated warning until it actually crashes.

ooeygui, 2020-04-17:
Waiting to hear back from the Azure Kinect SDK team on what could cause this. 
(Due to our WFH orders, I currently do not have the physical ability to replicate locally)

ooeygui, 2020-04-17:
Could you generate log files and post them?
https://docs.microsoft.com/en-us/azure/Kinect-dk/troubleshooting#collecting-logs

They have also pointed out this documentation on multi-kinect setups:
https://docs.microsoft.com/en-us/azure/kinect-dk/multi-camera-sync

wes-b, 2020-04-17:
Hi @carcormir, is all of this running in 1 process? The logs will be the best way to figure out what is happening. If the camera's are all in one process then only 1 log is needed. Also I see you are using k4aViewer this is a great tool for verifying your hardware. I encourage you to use the multi-camera-sync page to get your target configuration running without error first and then switching back to your application. And remember the Master must be the last sensor started.

tesych, 2020-04-17:
I will also recommend to review the [multicamera sync issue](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/530) on Sensor SDK GitHub

carcormir, 2020-04-18:
Hi @wes-b, thanks for your reply.
All of this is running under one ROS master node (process) and each camera driver is working under the master node and has its own node (process). I will get back with the log files as soon as possible, for the test with 3 cameras.

I wonder if for our purpose we need to wire the cameras. We basically have (for this application), 6 cameras in one PC and each camera is used at the same time but independenly. Could it be that the PC only has one USB controller and cannot handle multple devices at the same time?
If that is the case I do not understand why k4aviewer works fine even when creating 6 instances (one per camera) and ROS does not.

In any case, I will try to find some cables to connect the cameras and see how ROS responds.

Thanks for the inputs! I will come back with updates as soon as possible.

Have a great weekend.

wes-b, 2020-04-20:
It doesn't sound to me like you are really using our multi device synchronization. It sounds like you just have multiple camera's connected to 1 PC and are running them independently.

> Could it be that the PC only has one USB controller and cannot handle multiple devices at the same time?
> If that is the case I do not understand why k4aviewer works fine even when creating 6 instances (one per camera) and ROS does not.

I would expect that if you want to run with 6 cameras that you can validate that with 6 running instances of K4aViewer. I can't tell from what I have read if you have validate with them all running at the same time. Or just 1 at a time. 

When you applied https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/485, did you add enough memory for 6 camera's?

The logs will still be the most helpful. Something else that would be helpful is understanding what you are doing in ROS that triggers the start and stop of streaming.

ooeygui, 2020-04-21:
@carcormir I would like to find out more about your project. Can you email me at lamadio at microsoft dot com?

carcormir, 2020-04-22:
Hi @web-b, thanks for your reply.

> I would expect that if you want to run with 6 cameras that you can validate that with 6 running instances of K4aViewer. I can't tell from what I have read if you have validate with them all running at the same time. Or just 1 at a time.
> 
In fact that is how I validated them. 6 instances of k4aviewer running at the same time. This resulted in everything working fine.

> When you applied [microsoft/Azure-Kinect-Sensor-SDK#485](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/485), did you add enough memory for 6 camera's?
>
I did increase the memory to large upper bound to make sure there was memory for 6 cameras.

> The logs will still be the most helpful. Something else that would be helpful is understanding what you are doing in ROS that triggers the start and stop of streaming.
Sorry for my late answer, I could not get back to you faster. I hope to have some more details today (logs) and a small ROS package so you can try to replicate my issue.

Thanks for the help, and I will get back to you as soon as I can.

Kind regards,
Carlos

sibeichen, 2020-04-22:
Hi 
  I also had this problem with a camera

[2020-04-22 15:33:23.567] [error] [t=29041] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/color/uvc_camerareader.cpp (1300): DecodeMJPEGtoBGRA32(). MJPEG decode failed: -1
[2020-04-22 15:33:23.573] [error] [t=29042] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/queue/queue.c (112): queue_pop(). Queue "Queue_capture" was popped in a disabled state.
[2020-04-22 15:33:23.573] [error] [t=29042] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (269): capturesync_get_capture(device->capturesync, capture_handle, timeout_in_ms) returned failure in k4a_device_get_capture()
terminate called after throwing an instance of 'k4a::error'
  what():  Failed to get capture from device!
[ INFO] [1587540803.581669272]: Odom: quality=246, std dev=0.004126m|0.051274rad, update time=0.017660s
[k4a/manager-4] process has died [pid 28796, exit code -6, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=manager __log:=/home/bell/.ros/log/832bf016-846b-11ea-adfb-50eb7167c23a/k4a-manager-4.log].
log file: /home/bell/.ros/log/832bf016-846b-11ea-adfb-50eb7167c23a/k4a-manager-4*.log

carcormir, 2020-04-22:
Hi again @web-b,

**k4aviewer log**
Log files when running 3 instances of k4aviewer:
```
[ warning  ] : depth_engine_thread(). Dropping depth image due to bad timestamp at startup
[ warning  ] : depth_engine_thread(). Depth image processing is too slow at 1000ms (this may be transient).
```
That is basically what I get from the three devices. The warning occurs periodically and it is not persistent. It comes and goes. No errors.

**ROS logs**
These two folders contain all the information comming from ROS for each camera.
[test_cam1_fails.zip](https://github.com/microsoft/Azure_Kinect_ROS_Driver/files/4516309/test_cam1_fails.zip)
[test_cam2_fails.zip](https://github.com/microsoft/Azure_Kinect_ROS_Driver/files/4516310/test_cam2_fails.zip)
Notice that there are some irrelevant errors due to opening multiple cameras and the indexing of them, but those errors do not affect the performance. Basically the ROS driver package tries to open always the index 0 in the list no matter if the device is opened or not. If it is not available then tries index and so forth.

_test_cam2_fails/vision_2-manager-3-stdout.log_
Contains the main error message.
_test_cam1_fails/vision_1-manager-4-stdout.log_
Contains the main error message.
_rosout.log_
Contains all the information including how the other cameras are affected, like
`Couldn't find service manager/unload_nodelet, perhaps the manager is already shut down`

**Some extra information on the hardware.**
We have done some extra testing on two different PCs. These are the specs:
PC 1:
motherboard: ASRock x399M Taichi 
cpu: AMD Threadripper 1920X
gpu: ASUS RTX 2080Ti
PC 2:
motherboard: ASRock x399M Taichi 
cpu: AMD Threadripper 2970WX
gpu(2x): GIGABYTE RTX 2080Ti

Both computers have Ubuntu 18.04. and use the same kernel version.
PC 1: We are able to run 6 cameras on k4aviewer and ROS. No errors, just warnings like the one below in all cameras.
`[ WARN] [1586942399.828546544]: Initializing or re-initializing the device to realtime offset`
PC 2: We are able to run 6 cameras on k4aviewer but only 3 cameras with the ROS driver. Soon after the 3rd camera has been launch, the system breaks with the same error as always.

This is the launch file for the cameras. It is used as many times as cameras are launched but under different namespaces.
[cameralaunch.txt](https://github.com/microsoft/Azure_Kinect_ROS_Driver/files/4516425/cameralaunch.txt)
To run just use this line as many times as cameras are being used. Change sensor and sensor_sn parameters depending on the camera that is being launched:
`roslaunch <your_package_name> cameralaunch.launch sensor:=vision_1 sensor_sn:=000xxxxxx #remember to change the .txt to .launch`

wes-b, 2020-04-22:
Sorry @carcormir, first time I am working with a customer of customer, so we have a bit of learning to do in asking better questions. None of the logs uploaded contained:

> [ warning  ] : depth_engine_thread(). Dropping depth image due to bad timestamp at startup

This log message looks like it might have been copy and pasted from the K4AViewer rather than from the ROS run. I was looking for K4A logs documented here: https://docs.microsoft.com/en-us/azure/Kinect-dk/troubleshooting#collecting-logs, or K4A logs merged with ROS logs via [this API](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/group___functions_ga77a5f83e56a04136df1398478353193d.html#ga77a5f83e56a04136df1398478353193d) while you were reproducing the failure. Is it possible it is still in one of those many log files you shared?

If we collect logs right I would be able to find the error messages about depth_engine_thread or bad queue state.

Are you running Body Tracking? I see BT code in the repo that may or may not be compiled in (K4A_BODY_TRACKING). BT would have a hard time running 6 instances as it uses A LOT more compute resources then just the K4A Sensor SDK. Older versions of BT don't support multiple instances so we would need to figure out what BT version you are using.

carcormir, 2020-04-24:
Hey @wes-b,

> This log message looks like it might have been copy and pasted from the K4AViewer rather than from the ROS run.
It was copy and pasted as you said since the full log files only contained that message for each camera (next time I will post the entire file and not just a line, sorry for that). The warning message showed when running the cameras on K4AViewer and not ROS. I just wanted to show you a comparison of those logs and the ones I get from ROS.

> K4A logs merged with ROS logs via this API while you were reproducing the failure
I will look into this. Currently the only ROS that is being used is launching the drivers which run different nodelets to rectify images and to create the K4A-ROS nodelet bridge. Is there any feature to activate k4a.logs when launching the camera in ROS?

> Is it possible it is still in one of those many log files you shared?
I checked and those warnings do not appear in the log files I sent you.
As I mentioned the most informative information I found in those files is in _test_cam2_fails/vision_2-manager-3-stdout.log_, _test_cam1_fails/vision_1-manager-4-stdout.log_ and _rosout.log_

> Are you running Body Tracking? 
We are not using body tracking, only the K4A sensor SDK.

Thanks again for the help!

carcormir, 2020-04-24:
Hi @wes-b, more updates:

- We replaced the SSD drive from the PC that everything works (and I mentioned in one of my previous comments) to the PC that does not work. This did not change the behavior. Still same error. For that we assume is not an issue with the OS.
- We have attached a USB controller card with 4 independent controllers. When attaching 3 cameras to this card and opening 3 instances of k4aviewer, we get the error that we normally get only in ROS. For that we assume is not an issue with ROS.
These are the 3 log files from the 3 cameras launched with k4aviewer. This is the order that they were launched. camera3 is the one failing. camera1 and camera2 were manually stopped after the error came up, as you may see in the log files.
[camera1_k4a.log](https://github.com/microsoft/Azure_Kinect_ROS_Driver/files/4528497/camera1_k4a.log)
[camera2_k4a.log](https://github.com/microsoft/Azure_Kinect_ROS_Driver/files/4528498/camera2_k4a.log)
[camera3_k4a.log](https://github.com/microsoft/Azure_Kinect_ROS_Driver/files/4528499/camera3_k4a.log)

I hope this can help you guy get things a bit more clear.

Thanks in advance for the help!

wes-b, 2020-04-24:
Camera3 is the only camera that shows an error:

> [ error    ] ( /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/color/uvc_camerareader.cpp:1300 ) : DecodeMJPEGtoBGRA32(). MJPEG decode failed: -1

Does that somehow trigger the other cameras/node to stop? It sounds be like you were saying they run independently. (Remember I don't know ROS)

The error above indicates memory corruption in frame data. This typically happens in the application process space or can happen in hardware. It doesn't happen over USB due to the protections the USB protocol offer. Because you just repro'd this with K4AViewer, I am thinking you might have a bad camera. Can you identify the camera causing the failure and retest?

carcormir, 2020-04-24:
Hi @wes-b,
They do run independently in ROS and I guess also in K4aviewer when having 3 different instances.
But for the current testing, I am not using ROS at all.

> Can you identify the camera causing the failure and retest?
I did and retested the same way as before. 3 instances of K4aviewer. At some point one or more cameras rise the same error message and it is not always the same one. If I set to the maximum rgb resolution and FPS, the error appears faster.

wes-b, 2020-04-24:
Thanks, I have repro'd the issue with 3 instances of K4aViewer. I captured a potential workaround there, not sure if you can avoid using BGRA32.

https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1187

carcormir, 2020-04-27:
Thanks for the extra information @wes-b 

I launched 3 instances of K4aViewer but now with MJPEG format so the decoding to BGRA is avoided.
Unfortunately it did not help. No error appears in the log but K4aviewer pops up an error on screen.

`Color Camera: Received malformed image data`

Below you may find the error occurring in 2 of the 3 K4aviewer instances that are running.

[image]

wes-b, 2020-04-27:
K4aViewer appears to use the same tjDecompress2 function that the SDK uses when mjpeg is selected. You could try 1 of the other 2 formats, they rely on a different library to do the conversion.

carcormir, 2020-04-27:
That would be great but the problem is that the Azure Kinect ROS driver only allows you to choose BGRA or JPEG. Maybe @ooeygui has some inputs on this?

ooeygui, 2020-04-27:
@wes-b, @carcormir, we can add a [feature to support mjpg](https://github.com/microsoft/Azure_Kinect_ROS_Driver/blob/f953f9196e07fd7660aa29411c9e69445513eaef/src/k4a_ros_device.cpp#L94). The reason this hasn't been supported is because BGRA is used to construct the pointcloud2 message, so it relies on the SDK to do the translation. If the SDK isn't translating the frames, then we would have to do that manually in the ROS node.

(Note, the composition to RGBD is not accelerated which has caused issues from some customers. We have a workstream scheduled in the summer to address that)

@carcormir - Do you need the RGB data or can you work with just depth?

carcormir, 2020-04-27:
@ooeygui I believe that what @wes-b is suggesting is to give support for NV12 or YUY2 which do not rely on the tjDecompress2 function which is the one that is causing the issues when selecting BGRA or MJPEG in K4aviewer. So I do not think that giving ROS support for MJPEG  would solve the issue since it comes from the SDK and this issue will be reflected in the ROS driver.

@ooeygui We do use pointcloud, RGB and also the depth information (in the RGB frame as well as in the depth frame).

ooeygui, 2020-04-27:
@carcormir, Thank you for clarifying - I missed that in the thread.

wes-b, 2020-04-27:
I am definitely not suggesting any ROS changes yet, just offering ways to potentially workaround the problem. We unfortunately don't know enough about the issue yet.

carcormir, 2020-04-27:
@wes-b Please do not hesitate to contact me if there is any way I can help or if you have any updates on the issue.

carcormir, 2020-04-28:
Hi @wes-b, sorry to bother you again but we are really trying to solve this issue since soon it will interfere with our workflow.

The error comes from line 1288 in [uvc_camerareader.cpp](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/master/src/color/uvc_camerareader.cpp). In that code, the status of the decompression is only valid if it is 0. Otherwise it will return an error with no further checking.
In this issue [Cannot open MJPEG image coming from Camera
#157](https://github.com/libjpeg-turbo/libjpeg-turbo/issues/157) they mention that tjDecompress2 may return -1 if there is an error or a warning. They suggest to use a function tjGetErrorCode() to check if the error was a result of a warning or a fatal error.

I would suggest you to investigate if it is possible to implement this in the SDK since the error that I am getting could potentially be a warning from tjDecompress2. From the external libraries that are used in the SDK, I can see that tjGetErrorCode().

wes-b, 2020-04-28:
Hi @carcormir, thanks for this great info. I hope you don't mind but we should move this conversation over to Azure Kinect SDK so other members of our community can benefit.

https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1187

closed 2020-06-02 by carcormir

AswinkarthikeyenAK, 2020-08-24:
Hi @ooeygui ,
I am facing the same issues as discussed here below.
Have you already found a way to fix this issue? Can you kindly tell me how to fix this issue?

> Hi @ooeygui, thanks for your response. We are looking forward to hearing from you.
> In the meantime, we keep trying different things to figure out the root of the issue. I will keep you posted if we make any progress.
> 
> Just another note to my previous description. Before the error rises, I constantly get this message:
> `[ WARN] [1586942399.828546544]: Initializing or re-initializing the device to realtime offset`
> I do not know if that can be an indicative of something but it is a repeated warning until it actually crashes.
> 
> Again, thanks for the help!