microsoft/Azure_Kinect_ROS_Driver #126: Support Multi-camera setup

ooeygui, 2020-04-09:
We have a number of requests for multi-camera setup support. While it has been plumbed through, it doesn't work in all cases. This is a task to make it work and a supported scenario.

We will be duplicating all efforts to this task.

AswinkarthikeyenAK, 2021-07-13:
Hi @ooeygui,
Is there any updates regarding this feature? 
Thanks

ooeygui, 2022-07-18:
Hi @jyang-cpr,
This item is on our backlog of ROS and robotics work. When this work gets funded, the ROS distro determination will depend on the funding project requirements. I do not have a time horizon for a 1st party implementation. I would be happy to support a community contribution to this effort, including design questions and PRs.