microsoft/Azure_Kinect_ROS_Driver #128: Camera_info switched in for rect image pub

tanyouliang95, 2020-04-28:
Hi, this is just regarding about the ros camera_info data pub issue. I have noticed that the camera_info is switched in here while publishing it with the rectified topics. 
```cpp
  calibration_data_.getDepthCameraInfo(rgb_rect_camera_info);
  calibration_data_.getRgbCameraInfo(depth_rect_camera_info);
```
Refer to [here](https://github.com/microsoft/Azure_Kinect_ROS_Driver/blob/f953f9196e07fd7660aa29411c9e69445513eaef/src/k4a_ros_device.cpp#L812)

While the raw images topics are publishing the "correct" camera_info, the rectified images seems to be switched. I supposed it should be the other way as such
```cpp
  calibration_data_.getDepthCameraInfo(depth_rect_camera_info);
  calibration_data_.getRgbCameraInfo(rgb_rect_camera_info);
```
I am also quite unsure, please correct me if I am wrong with this.

ooeygui, 2020-04-28:
It looks like this was an oversight during the implementation. Good catch!
How does this bug manifest for you?

tanyouliang95, 2020-04-29:
This is not significant to people who directly subcribe to the published pointcloud. For me, it doesnt affect my application. But if anyone is planning to use the `camera_info` of rect images, this will be confusing to them. Cheers!

christian-rauch, 2021-02-04:
I am using the camera intrinsics to do the projection myself. Since the sensor is factory calibrated, there is only a single set of intrinsic parameters per camera (`{rgb,ir}/camera_info`). The remaining intrinsics should be derived from the registration into the respective camera frames:
- **depth ← ir** (depth is derived from the IR image)
- **depth_to_rgb ← rgb** (depth data is assigned to pixels in the colour frame)
- **rgb_to_depth ← rgb** (colour data is assigned to pixels in the depth/ir frame)

The `rgb/camera_info` and `ir/camera_info` are correctly assigned from `getRgbCameraInfo` and `getDepthCameraInfo` respectively.

`rgb_to_depth`, which registers the colour in the depth frame (assigns colour to depth data), uses `getDepthCameraInfo` which is correct. `depth_to_rgb`, which registered the depth in the colour frame (assigns depth to the colour data), uses `getRgbCameraInfo` which is also correct.

All this seems correct. But I agree that the "rect" variable naming is very confusing because `depth_rect_camera_info` is actually the calibrated RGB image and `rgb_rect_camera_info` is actually the calibrated IR image.

You could simply assign the "rect" intrinsics from the original `rgb` and `ir` intrinsics, leaving a comment why this is done.

@ooeygui Any chance that his variable naming gets fixed?