microsoft/Azure_Kinect_ROS_Driver #129: When will the ROS 2-dashing version of azure Kinect be released?

KevinHwangAi, 2020-04-30:
Hi Engineers of azure Kinect.

Our team is preparing to use azure Kinect as the main depth camera in our robot project. 
Currently, we want to only use ros2 dashing as the main system tool.
Since last year, we have been looking forward to the support of ROS2-Dashing version from azure Kinect.Could you tell us when will the ROS 2-dashing version of azure Kinect be released?

ooeygui, 2020-04-30:
Hello @KevinHwangAi ,
I'd love to know more about your project.
We currently have ROS2 support for all our first party ROS nodes scheduled for June. 

Our plan is to enable the first party nodes for Foxy forward.

KevinHwangAi, 2020-05-01:
Hi @ooeygui .
Thank you for your answer. We are looking forward to it. 

For our project, we hope to control the robot through Kinect motion capture.

ooeygui, 2020-05-01:
That sounds like a great project.

Have you investigated using the ROS1 bridge in the mean time?

KevinHwangAi, 2020-05-01:
Not yet.Maybe we will try later.Thanks.

tkircher, 2020-05-03:
@KevinHwangAi Our team went with the RealSense D435 + T265 over the Kinect or Zed 2 because of Intel's better support. Still keeping my fingers crossed for the Kinect though, since it's the only real LIDAR camera until the L515 ships.

KevinHwangAi, 2020-05-03:
@tkircher ,Yes, currently only Kinect provides TOF camera.That's why I want to use Kinect for development.

KevinHwangAi, 2020-05-06:
Hi @ooeygui.
I am testing the  ros1 bridge to connect Kinect and ros2, but the result is not very satisfactory.The depth image with rviz2 has a large delay,so unable to capture action in real time.

ooeygui, 2020-10-30:
Working on the Foxy builds now
https://github.com/microsoft/Azure_Kinect_ROS_Driver/pull/155

francescosemeraro, 2021-03-31:
Good afternoon @ooeygui.
Has there been any progress about the Foxy builds for Azure Kinect? I am trying to make the Azure Kinect ROS Driver in ROS 2, but it is becoming very painful to do it on my own. Having a version easily deployable on ROS 2 could really help me a lot, thanks.

ooeygui, 2021-03-31:
@francescosemeraro Foxy builds have been committed as part of https://github.com/microsoft/Azure_Kinect_ROS_Driver/pull/155; are you seeing issues with them?

closed 2021-03-31 by ooeygui