microsoft/Azure_Kinect_ROS_Driver #134: capturesync_drop Error on occasion

tanyouliang95, 2020-06-02:
Running on Ubuntu 18.04
Azure Kinect SDK: v1.4

During the running of the k4a_ros node, this issue happens quite sporadically, 
```
[2020-06-02 17:45:45.810] [error] [t=1202] ../src/capturesync/capturesync.c (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    594122 type:Depth
[2020-06-02 17:45:46.010] [error] [t=1202] ../src/capturesync/capturesync.c (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    786877 type:Depth
[2020-06-02 17:45:46.210] [error] [t=1202] ../src/capturesync/capturesync.c (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    986877 type:Depth
[2020-06-02 17:45:46.410] [error] [t=1202] ../src/capturesync/capturesync.c (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1186877 type:Depth
```

When this error msg happens, it is not possible to recover, in which a forceful restart of the node is needed. As error will happen occasionally, I not able to reproduce it. 

According to some previous this raised issue: 
https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1112

This might be due to the gpu resources. However sometimes it will also happen while the gpu usage is still low. Is there anyway to solve this? or maybe raise an exception under the ros_driver when this happen, to make sure the node is able to recover soon after the TS is back in synced?

The Ros_driver node is stable most of the time, but sometimes these unexpected errors will kick in randomly.

ooeygui, 2020-06-11:
Can you post the specs of the system you are running on?

ooeygui, 2020-06-24:
Reopen if you can add more info

closed 2020-06-24 by ooeygui

tanyouliang95, 2021-01-28:
Just some long due reply on this issue. This is running on `ubuntu18.04`. Previously after trying out with different azure kinect devices, the issue seems to only happen to one device. This seems to solve the issue using a different azure kinect.