microsoft/Azure_Kinect_ROS_Driver #143: find_dependency(k4a) version format mismatch (contains 3, requested 2 components)

anastasiabolotnikova, 2020-07-19:
Running `catkin_make` to build `Azure_Kinect_ROS_Driver` results in the following CMake error:
```bash
Finding K4A SDK binaries
CMake Error at Azure_Kinect_ROS_Driver/cmake/Findk4a.cmake:22 (message):
  Error: Azure Kinect SDK Version numbers contain exactly 3 components
  (major.minor.rev).  Requested number of components: 2
Call Stack (most recent call first):
  /usr/local/share/cmake-3.18/Modules/CMakeFindDependencyMacro.cmake:47 (find_package)
  /usr/lib/cmake/k4abt/k4abtConfig.cmake:3 (find_dependency)
  Azure_Kinect_ROS_Driver/cmake/Findk4abt.cmake:35 (find_package)
  Azure_Kinect_ROS_Driver/CMakeLists.txt:94 (find_package)

-- Configuring incomplete, errors occurred!
```
Looking into the file that seems to cause the problem `cat /usr/lib/cmake/k4abt/k4abtConfig.cmake` I see indeed only 2 components in the version number:

```cmake
include(CMakeFindDependencyMacro)

find_dependency(k4a 1.3 REQUIRED)

# Add the targets file
include("${CMAKE_CURRENT_LIST_DIR}/k4abtTargets.cmake")
```

Changing this file manually to include 3 version number components:

```cmake
find_dependency(k4a 1.3.0 REQUIRED)
```

fixes the problem and `Azure_Kinect_ROS_Driver` can be built successfully with `catkin_make`. 

**To Reproduce**
1. `sudo apt install k4a-tools=1.3.0`
1. `sudo apt install libk4a1.3-dev`
1. `sudo apt install libk4abt1.0-dev`
1. `sudo apt-mark hold k4a-tools`
1. `git clone git@github.com:microsoft/Azure_Kinect_ROS_Driver.git catkin_ws/src`
1. `cd catkin_ws`
1. `catkin_make`

**Expected behavior**
`catkin_make` allows to successfully build `Azure_Kinect_ROS_Driver` without the necessity to manually change `/usr/lib/cmake/k4abt/k4abtConfig.cmake`

**Desktop:**
 - OS: Ubuntu 18.04.4 LTS
 - CMake: [3.18](https://github.com/Kitware/CMake) at [fa7157b](https://github.com/Kitware/CMake/commit/f74a963021b1ca26c7145f53a622183522108bdf)

maxsvetlik, 2020-09-14:
I haven't experienced this issue with CMake, but I was facing issues with `k4a-tools` having a conflicting version with the libraries needed for this ROS driver, which is addressed in your reproduction steps.

Rather than downgrading `k4a-tools` and preventing its update with apt, I've found that I you can get by without using `k4a-tools` at all (at least in my use case). So just installing the Kinect Body Tracking SDK with `sudo apt install libk4abt1.0-dev` installs the correct libraries for use with this ROS wrapper. So steps 1 and 4 in your reproduction list could be eliminated.

xiaobiaozhu, 2021-08-29:
That helps a lot! Thanks.

Vins9651, 2021-12-13:
After having spent hours trying to understand what the problem could be, finally I've found this solution and solved it immediately.
Thanks a lot!!

HaooWang, 2021-12-24:
Good job,  I have the same problem, that helps a lot!!!! Thank you .