microsoft/Azure_Kinect_ROS_Driver #144: Publish body markers array only when body markers are detected

anastasiabolotnikova, 2020-07-22:
**Is your feature request related to a problem? Please describe.**
When no body is detected, an empty markers array is being published to the `body_tracking_data` topic. This requires any node, subscribed to this topic, to write an additional check in the subscriber callback to see if marker array has any elements. It would be more convenient (and ROS-like) to only publish messages to `body_tracking_data` when at least one body is detected and markers array is not empy.

**Describe the solution you'd like**
Publish body markers array only when body markers are detected. 
In `framePublisherThread` add `if(num_bodies > 0)` check, only then publish:
```c++
if(num_bodies > 0){
  body_marker_publisher_.publish(markerArrayPtr);
}
```

**Describe alternatives you've considered**
Every time a node subscribes to a `body_tracking_data`, it has to implement a check in the subscriber callback to see if the body marker array is non-empty. This is quite tedious for the subscriber nodes, however. When I subscribe to a ROS topic, I expect to only receive meaningful messages (e.g. non-empty messages). In this case, I would expect to only receive messages on `body_tracking_data` topic, when body marker detection is successful (at least one body is detected).

RoseFlunder, 2020-07-22:
Hmm I kind of understand what you mean but isn't it also a meaningful information when you receive an empty message after a filled message? In this case the message would tell you that the person walked out of the picture / is occluded.

anastasiabolotnikova, 2020-07-22:
Thank you for your prompt reply, @RoseFlunder !
Fair point. However, this information (`no human detected`) can also be deduced from the subscriber callback function not being triggered at this time step (or at this control iteration, or for the past X seconds).

Previously, when I was working with ROS nodes that publish detection results, I recall that detection result messages were only published when detection was successful. You can see, for example, the [`aruco_ros`](https://github.com/pal-robotics/aruco_ros) package that does so: [only publishes detection if `markers.size()` is non-zero](https://github.com/pal-robotics/aruco_ros/blob/959fe5fcdc0148e358a180e0366b79bde08c74a8/aruco_ros/src/simple_single.cpp#L215). This is why, I was expecting the same behavior when using `Azure_Kinect_ROS_Driver`. 

I agree that it could be an open question, whether this check should be done before publishing the message in `Azure_Kinect_ROS_Driver` or it should be left to the user to implement it in their subscriber node callbacks. 
We can see if more people joint this discussion and share their preference regarding this matter.

ooeygui, 2022-04-19:
I believe this has been implemented. Please reopen if this isn't what you expect

closed 2022-04-19 by ooeygui