microsoft/Azure_Kinect_ROS_Driver #158: Ros Body tracking - Joint information and visualization of the marker array[question]

AswinkarthikeyenAK, 2020-11-12:
Hi, 
Is there a way to visualize the joint data as TF's in RViz?
And also I made changes to the k4a_ros_device.cpp file to change the type of the marker to CUBE and launched the driver.launch again. But was not able to notice any changes.
Can you please tell me how can I change this to cube?
`  marker_msg->lifetime = ros::Duration(0.25);
  marker_msg->id = body.id * 100 + jointType;
  marker_msg->type = Marker::CUBE;`

RoseFlunder, 2020-11-12:
1. Question:
I think you would have to rewrite the whole body tracking publishing thing to tf Message publishers per joint using the ROS tf2 classes.

2. Question:
If you just want to change the marker type your change seems right. Did you build the catkin workspace after changing the line?

AswinkarthikeyenAK, 2020-11-12:
Thanks for your reply.

2. Sorry, I did not catkin make therefore I believe it did not work. I did catkin_make and it work now. Thanks for that

1. I am new to ROS and TF, So i will take a look at the link you have shared and see if I can write whole body tracking publishing to send the TF messages. Any information or assistance that you can provide is highly appreciated.  

Thanks

RoseFlunder, 2020-11-12:
Basically you need to publish each body tracking frame new transformations.
The TFMessage contains an array of stamped transformations (stamped just mean they are related to a specific point in time).

A transformation has a parent frame id in the header field (name of the parent coordinate system) and a child frame id which is the name of the coordinate system you want to transform to.
So you span a tree of transformations with all the messages.

For example the first transform would be from the depth camera coordinate system to the skeleton root joint (pelvis in this case because it has no other parent joints).
The next message would describe a transformation from the coordinate system from the pelvis joint to one of its child joints, i.e. spine naval etc. etc.

The transformation itself has two parts.
The translation part, so how much you have to move the parent coordinate systems origin in X-, Y-, Z-direction to get to the childs coordinate system origin.
The second part is the rotation part, so how you have to rotate the coordinate systems axis from the parent coordinate system to be in line with childs coordinate system.

ooeygui, 2022-04-19:
This has been implemented in ROS1 and ROS2 is in progress.

closed 2022-04-19 by ooeygui