microsoft/Azure_Kinect_ROS_Driver #173: catkin_make Azure_Kinect_ROS_Driver error

xspkwy, 2021-01-20:
- **Description**: I have followed the official guide of **_Azure-Kinect-Sensor-SDK_** and test k4aviewer successfully, but when I then created a ros workspace, downloaded **_Azure_Kinect_ROS_Driver_** and catkin_make it, it seems to fail:
`Scanning dependencies of target azure_kinect_ros_driver_nodelet
Scanning dependencies of target azure_kinect_ros_driver_node
[ 10%] Building CXX object Azure_Kinect_ROS_Driver/CMakeFiles/azure_kinect_ros_driver_node.dir/src/k4a_ros_bridge_node.cpp.o
[ 20%] Building CXX object Azure_Kinect_ROS_Driver/CMakeFiles/azure_kinect_ros_driver_node.dir/src/k4a_ros_device.cpp.o
[ 30%] Building CXX object Azure_Kinect_ROS_Driver/CMakeFiles/azure_kinect_ros_driver_node.dir/src/k4a_ros_device_params.cpp.o
[ 40%] Building CXX object Azure_Kinect_ROS_Driver/CMakeFiles/azure_kinect_ros_driver_nodelet.dir/src/k4a_ros_bridge_nodelet.cpp.o
[ 50%] Building CXX object Azure_Kinect_ROS_Driver/CMakeFiles/azure_kinect_ros_driver_nodelet.dir/src/k4a_ros_device.cpp.o
[ 60%] Building CXX object Azure_Kinect_ROS_Driver/CMakeFiles/azure_kinect_ros_driver_nodelet.dir/src/k4a_ros_device_params.cpp.o
[ 70%] Building CXX object Azure_Kinect_ROS_Driver/CMakeFiles/azure_kinect_ros_driver_node.dir/src/k4a_calibration_transform_data.cpp.o
[ 80%] Building CXX object Azure_Kinect_ROS_Driver/CMakeFiles/azure_kinect_ros_driver_nodelet.dir/src/k4a_calibration_transform_data.cpp.o
[100%] Linking CXX executable /home/vclabr/KinectDK_ws/devel/lib/azure_kinect_ros_driver/node
[100%] Linking CXX shared library /home/vclabr/KinectDK_ws/devel/lib/libazure_kinect_ros_driver_nodelet.so
[100%] Built target azure_kinect_ros_driver_nodelet
CMakeFiles/azure_kinect_ros_driver_node.dir/src/k4a_ros_device.cpp.o：在函数‘cv::Mat::Mat(int, int, int, void*, unsigned long)’中：
k4a_ros_device.cpp:(.text._ZN2cv3MatC2EiiiPvm[_ZN2cv3MatC5EiiiPvm]+0x296)：对‘cv::Mat::updateContinuityFlag()’未定义的引用
collect2: error: ld returned 1 exit status
Azure_Kinect_ROS_Driver/CMakeFiles/azure_kinect_ros_driver_node.dir/build.make:214: recipe for target '/home/vclabr/KinectDK_ws/devel/lib/azure_kinect_ros_driver/node' failed
make[2]: *** [/home/vclabr/KinectDK_ws/devel/lib/azure_kinect_ros_driver/node] Error 1
CMakeFiles/Makefile2:519: recipe for target 'Azure_Kinect_ROS_Driver/CMakeFiles/azure_kinect_ros_driver_node.dir/all' failed
make[1]: *** [Azure_Kinect_ROS_Driver/CMakeFiles/azure_kinect_ros_driver_node.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed`

- I have searched for long and found two similar issues: [issue #127](https://github.com/microsoft/Azure_Kinect_ROS_Driver/issues/127) and [issue #1135](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1135)
I have tried their solutions, and checked my OpenCV version by opencv_version and it shows 3.4.7, which confuses me because they said the issues may due to the version of opencv is too old.

- System config: ubuntu18.04, ros-melodic, libk4a1.4 libk4a1.4-dev k4a-tools, opencv3.4.7 which is built from source.

 So if any applicable solutions? Thanks!

ooeygui, 2021-01-20:
Thank you for the report. The Interface for OpenCV is different between versions. A code `#ifdef` needs to be added to detect the OpenCV version, and compile for the different interface.

We would appreciate a PR for this.

xspkwy, 2021-01-21:
Thanks for your advice, I would submit a PR when I fixed the issue. But I am still not sure how to fix it, could you tell me the details of changes?

ooeygui, 2021-01-21:
Hello @xspkwy , 
Thank you for your interest in looking at the PR.
OpenCV headers provide a version define `CV_MINOR_VERSION` & `CV_MINOR_VERSION`. You can check them like this:
``` c++
#if (CV_MAJOR_VERSION > 3 ||  (CV_MAJOR_VERSION == 3 && CV_MINOR_VERSION >= 7))
 // new code, assuming it also compiles on OpenCV 4+
#else
// older code
#endif
```