microsoft/Azure_Kinect_ROS_Driver #175: compressed depth is in wrong format

christian-rauch, 2021-01-26:
**Describe the bug**
The compressed depth on topics `/depth/image_raw/compressed` and `/depth_to_rgb/image_raw/compressed` uses the jpeg format by default, which is 1) not a good format to compress depth data, and 2) results in errors:
`Compressed Image Transport - JPEG compression requires 8/16-bit color format (input format is: 32FC1)`

Changing this to "png" does not help.

In addition, the alternative topic `/depth/image_raw/compressedDepth` is only published at 3Hz instead of 30Hz like the raw uncompressed depth.

**To Reproduce**
Steps to reproduce the behavior:
1. start driver: `roslaunch azure_kinect_ros_driver driver.launch`
2. visualise images: `rqt_image_view /depth/image_raw/compressed
3. see error: `Compressed Image Transport - JPEG compression requires 8/16-bit color format (input format is: 32FC1)`

**Expected behaviour**
The ROS node should publish valid messages for the compressed depth images.

**Desktop (please complete the following information):**
 - OS: Ubuntu
 - Version: 20.04

RanLongUoE, 2021-02-03:
I faced the exact same problem when I tried to record "/depth_to_rgb/image_raw/compressed". I have tried the solution in https://github.com/microsoft/Azure_Kinect_ROS_Driver/issues/67, but it didn't help.
Desktop:
    OS: Ubuntu
    Version: 18.04

christian-rauch, 2021-02-03:
As a reference, the [`openni2_camera`](http://wiki.ros.org/openni2_camera) driver publishes on `/camera/depth/image` in float (`32FC1`, meter) format and on `/camera/depth/image_raw` in integer (`16UC1`, 1 unit = 1 meter) format. The png compression can then be applied to the `image_raw` topic in `16UC1` without issues.

The Kinect driver only publishes the floating `32FC1` values on `/depth/image_raw`. I assume there should be a topic `/depth/image` in metric units, and topic `/depth/image_raw` should have the integer `16UC1` units that can be compressed using the png compression. The benefit of the `16UC1` png compression is that this can be easily used in python, as the float compression on `compressedDepth` is not supported in puthon at the moment.

PR #180 (2021-07-12): publish compressed depth in 16bit PNG format
## Fixes #175 

### Description of the changes:
- change default depth units to 16bit millimetre
- change default `compressed/format` for depth topics to `png`
- set lowest (fastest) compression level

In total, these changes enable the publishing of compressed depth images at 30Hz.

### Required before submitting a Pull Request:
- [x] I reviewed [CONTRIBUTING.md](https://github.com/microsoft/Azure_Kinect_ROS_Driver/blob/master/CONTRIBUTING.md)
- [x] I [built my changes](https://github.com/microsoft/Azure_Kinect_ROS_Driver/blob/master/docs/building. [... cut]
PR #203 (2021-07-28): Declare depth params
## Fixes #175

### Description of the changes:
- These are additional fixes for issue #175 that were addressed in PR #180 for ROS 2
- Added declarations for depth related topics as these parameters cannot be set before they are declared. 

### Required before submitting a Pull Request:
- [x] I reviewed [CONTRIBUTING.md](https://github.com/microsoft/Azure_Kinect_ROS_Driver/blob/master/CONTRIBUTING.md)
- [x] I [built my changes](https://github.com/microsoft/Azure_Kinect_ROS_Driver/blob/master/docs/building.md) locally
- [x] I tested my changes with a device
- [ ] I changed b [... cut]
closed 2021-07-12 by amelhassan