microsoft/Azure_Kinect_ROS_Driver #182: Body tracking data publishing frequency

AswinkarthikeyenAK, 2021-03-16:
**Describe the bug**
I am trying to use the body tracking feature to get various joint information. The camera is set at 30 fps, and body tracking smoothing factor is set to 0.75. However the joint informations are published at approx 5.5 Hz. The rest of the topics for example /depth_to_rgb/image_raw seems to be publishing information at 30 Hz. 
For Example:
```
subscribed to [/depth_to_rgb/image_raw]
average rate: 30.864
	min: 0.009s max: 0.042s std dev: 0.00571s window: 28
average rate: 30.468
	min: 0.009s max: 0.042s std dev: 0.00422s window: 58
average rate: 30.324
	min: 0.009s max: 0.042s std dev: 0.00369s window: 88
average rate: 30.268
	min: 0.009s max: 0.042s std dev: 0.00341s window: 118
average rate: 30.194
	min: 0.009s max: 0.042s std dev: 0.00323s window: 148
average rate: 30.178
	min: 0.009s max: 0.042s std dev: 0.00304s window: 178
average rate: 30.122
	min: 0.009s max: 0.044s std dev: 0.00301s window: 208
average rate: 30.092
	min: 0.009s max: 0.044s std dev: 0.00292s window: 238
^Caverage rate: 30.107
```

```
subscribed to [/body_tracking_data]
average rate: 5.785
	min: 0.165s max: 0.185s std dev: 0.00787s window: 5
^Caverage rate: 5.731
	min: 0.164s max: 0.187s std dev: 0.00796s window: 10

```
**To Reproduce**
Steps to reproduce the behavior:
1. `roslaunch azure_kinect_ros_driver driver.launch` with body tracking enabled
2. ` rostopic hz /body_tracking_data`

**Expected behavior**
Shouldn't the /body_tracking_data should publish data at 30 Hz or Is this the maximum frequency at which the /body_tracking_data will be publishing the data? 

**Desktop (please complete the following information):**
 - OS: Ubuntu
 - Version 18.04
 - ROS Melodic

**Additional context**
Add any other context about the problem here.

ooeygui, 2021-03-16:
Hello @AswinkarthikeyenAK,
The ROS node for the Azure Kinect is a thin wrapper around the Azure Kinect SDK to bridge the SDK into ROS. 

The Azure Kinect SDK team is working on Body Tracking performance and has outstanding tasks tagged "Body Tracking":
https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues?q=is%3Aissue+is%3Aopen+label%3A%22Body+Tracking%22

closed 2021-03-16 by ooeygui

AswinkarthikeyenAK, 2021-03-18:
Hi @ooeygui ,
I see.. Does this mean the its the issue with the Body-tracking sending data at slow rate?

Thanks

RoseFlunder, 2021-03-18:
Most likely because there is no artificial throtteling for body tracking data in this node.
What GPU are you using and do you have other topics subscribed as well?
RGB point cloud can be so slow that it slows down the whole processing loop.

AswinkarthikeyenAK, 2021-03-18:
Hi @RoseFlunder ,
I have a Nvidia MX130 graphic card. 
I am not subscribing to any other topics. I am just working with the /body_tracking_data

Thanks

RoseFlunder, 2021-03-18:
The MX130 is unfortunately too slow to keep up with the 30FPS.
Microsoft recommends at least a GTX 1070 for 30 FPS.
I used a GTX 1080 without problems.
But be aware that there is currently no support for the new Nvidia 3000 Series.
Should be patched with the next release.

ooeygui, 2021-03-18:
@roseflunder Thank you for posting on this issue.

AswinkarthikeyenAK, 2021-03-24:
Hi @RoseFlunder, 
Is there any estimate on when the support might be extended to the Nvidia 3000 series? 
I noticed there already an update in [v1.1.0](https://docs.microsoft.com/en-us/azure/kinect-dk/body-sdk-download) that supports the RTX 30xx series. 
Is there a way you can check if the 30xx series works fine with ROS as well please?
I am planning to buy a new device with the RTX 30xx, but it would be really helpful if you can confirm it works fine with the ROS Body tracking without any issues.

Thanks

RoseFlunder, 2021-03-24:
Hi,
I recently finished my studies at university where I worked with azure kinect bodytracking, which means I don't have access to a device anymore.
So I can't test/develop any new things for this ros driver anymore.
If there are no breaking API changes you could test to change the version number of k4a und k4abt dependencies in the build files and test if it compiles without any further changes.
Additional things like specifying a model path would need some changes to the code though.

AswinkarthikeyenAK, 2021-03-24:
Thanks, for your input.
Sure, i will give that a go. 

Thanks again