microsoft/Azure_Kinect_ROS_Driver #184: Asking for human body tracking on recorded rosbag files

maudzung, 2021-03-17:
I have used code in this repository to record data into rosbag files. Are there instructions or examples on loading the recorded rosbag files and doing human body tracking (skeleton extraction)? 

Thank you so much!

zchenpds, 2021-03-17:
k4abt tracker consumes data of type k4a::capture which cannot be easily reconstructed from the ros messages supplied by this node. Is there any reason you couldn't record the body tracking data into the bag file in the first place? or you want to try the body tracking with different smoothing factor settings?

maudzung, 2021-03-17:
Thanks for the information. I didn't record `body_tracking_data` topic. I'll record the topic next time for its usage.

closed 2021-03-17 by maudzung