microsoft/Azure_Kinect_ROS_Driver #186: catkin_make error

jiangSeu, 2021-03-25:
hello, 

when i  use the command catkin_make, an error was reported

CMake Error at Azure_Kinect_ROS_Driver/cmake/Findk4a.cmake:78 (find_package):
  Could not find a configuration file for package "k4a" that is compatible
  with requested version "1.3.0".

  The following configuration files were considered but not accepted:

    /usr/lib/x86_64-linux-gnu/cmake/k4a/k4aConfig.cmake, version: 1.1.1

Call Stack (most recent call first):
  Azure_Kinect_ROS_Driver/CMakeLists.txt:90 (find_package)

zchenpds, 2021-03-25:
You may want to upgrade your libk4a from version 1.1.1 to 1.3.0 following https://docs.microsoft.com/en-us/azure/kinect-dk/sensor-sdk-download#linux-installation-instructions
```bash
sudo apt install libk4a1.3-dev
```

jiangSeu, 2021-03-25:
thank you very much,  the issue is resolved!

closed 2021-03-31 by ooeygui