microsoft/Azure_Kinect_ROS_Driver #187: Issues in running colcon build for Azure ROS driver in ROS 2 in Ubuntu 20.04

francescosemeraro, 2021-03-31:
Good morning!

I am trying to build the Azure ROS Driver on ROS 2 in Ubuntu 20.04. When I run colcon build, I get the following error:

Starting >>> azure_kinect_ros_driver
--- stderr: azure_kinect_ros_driver
CMake Error at CMakeLists.txt:12 (find_package):
  By not providing "Findcatkin.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "catkin", but
  CMake did not find one.

  Could not find a package configuration file provided by "catkin" with any
  of the following names:

    catkinConfig.cmake
    catkin-config.cmake

  Add the installation prefix of "catkin" to CMAKE_PREFIX_PATH or set
  "catkin_DIR" to a directory containing one of the above files.  If "catkin"
  provides a separate development package or SDK, be sure it has been
  installed.

---
Failed   <<< azure_kinect_ros_driver [0.05s, exited with code 1]

Summary: 0 packages finished [0.14s]
  1 package failed: azure_kinect_ros_driver
  1 package had stderr output: azure_kinect_ros_driver

I thought I was missing catkin, so I also installed ROS Noetic. However, when I ran colcon build again, I got issues related to the ROS_DISTRO variable and it does not build my workspace. 
Could you please guide me through what I need to build this package in ROS 2 in Ubuntu 20.04?

ooeygui, 2021-03-31:
Are you using the Foxy-devel branch?

ooeygui, 2022-04-19:
This does not repro on the foxy-devel branch

closed 2022-04-19 by ooeygui