microsoft/Azure_Kinect_ROS_Driver #188: Body Tracking latest version k4a tools 1.4.1 and k4abt version 1.1.0 issue

AswinkarthikeyenAK, 2021-04-06:
**Describe the bug**
I tried to install the latest Kinect Azure SDK 1.4.1 and the body tracking SDK version 1.1.0.
When I launch the ROS node, I get the following error message.

```
2021-04-06 20:46:28.486403652 [E:onnxruntime:, sequential_executor.cc:333 Execute] Non-zero status code returned while running ArgMax node. Name:'ArgMax' Status Message: /tmp/onnxruntime/onnxruntime/core/framework/bfc_arena.cc:298 void* onnxruntime::BFCArena::AllocateRawInternal(size_t, bool) Failed to allocate memory for requested buffer of size 503316480

Non-zero status code returned while running ArgMax node. Name:'ArgMax' Status Message: /tmp/onnxruntime/onnxruntime/core/framework/bfc_arena.cc:298 void* onnxruntime::BFCArena::AllocateRawInternal(size_t, bool) Failed to allocate memory for requested buffer of size 503316480
```
and the body tracking does not seem to be working. 

**To Reproduce**
Steps to reproduce the behavior:
1. Install k4a-tools=1.4.1 and k4abt1.1 k4abt1.1-dev
2. roslaunch azure_kinect_ros_driver driver.launch

**Expected behavior**
```
2021-04-06 20:46:28.486403652 [E:onnxruntime:, sequential_executor.cc:333 Execute] Non-zero status code returned while running ArgMax node. Name:'ArgMax' Status Message: /tmp/onnxruntime/onnxruntime/core/framework/bfc_arena.cc:298 void* onnxruntime::BFCArena::AllocateRawInternal(size_t, bool) Failed to allocate memory for requested buffer of size 503316480

Non-zero status code returned while running ArgMax node. Name:'ArgMax' Status Message: /tmp/onnxruntime/onnxruntime/core/framework/bfc_arena.cc:298 void* onnxruntime::BFCArena::AllocateRawInternal(size_t, bool) Failed to allocate memory for requested buffer of size 503316480
```

**Desktop (please complete the following information):**
 - OS:  Ubuntu
 - Version 18.04

Can you kindly let me know how can fix the error and use the latest SDK for body tracking please?

ooeygui, 2021-04-06:
This looks like an SDK issue - are you able to reproduce this without the ROS node? (using the SDK tools directly)

AswinkarthikeyenAK, 2021-04-07:
Hi,
Without out the ROS node, it seems to be working. 
But I got something like this as soon as i started the k4abt_simple_3d_viewer
`Force fallback to CPU execution for node: Shape__40 `

zchenpds, 2021-04-07:
> But I got something like this as soon as i started the k4abt_simple_3d_viewer
`Force fallback to CPU execution for node: Shape__40 `

According to https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1555, the warning can be safely ignored.

Did you update the `CMakeLists.txt` file and `catkin_make`?
https://github.com/microsoft/Azure_Kinect_ROS_Driver/blob/cac73214020db1298decb6ad0d22aac45863a531/CMakeLists.txt#L89-L94

AswinkarthikeyenAK, 2021-04-07:
Hi, @zchenpds,
I did do that update the CMakeList.txt 
Thanks

HlibKazakov2000, 2021-04-09:
Hello, @AswinkarthikeyenAK 
I wasn't able to reproduce this issue.
Could you please describe the process of launching and setting up your environment command-by-command, so I can see the difference in our setups?
Thanks

HlibKazakov2000, 2021-04-12:
@AswinkarthikeyenAK 
Do you have enough free space to allocate the requested buffer?

ooeygui, 2022-04-19:
We have had recent improvements to the body tracking stack, closing this out.

closed 2022-04-19 by ooeygui