microsoft/Azure_Kinect_ROS_Driver #190: Build Error with OpenCV3.4.13

Shunichi09, 2021-04-27:
**Describe the bug**

I tried to build with opencv3.4.13 and I got the following errors.

```
CMakeFiles/azure_kinect_ros_driver_node.dir/src/k4a_ros_device.cpp.o: In function `K4AROSDevice::renderDepthToROS(boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > >&, k4a::image&)':
/home/sgrasp_robot/catkin_ws/src/Azure_Kinect_ROS_Driver/src/k4a_ros_device.cpp:394: undefined reference to `cv::Mat::Mat(int, int, int)'
CMakeFiles/azure_kinect_ros_driver_node.dir/src/k4a_ros_device.cpp.o: In function `K4AROSDevice::renderIrToROS(boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > >&, k4a::image&)':
/home/sgrasp_robot/catkin_ws/src/Azure_Kinect_ROS_Driver/src/k4a_ros_device.cpp:425: undefined reference to `cv::Mat::Mat(int, int, int)'
collect2: error: ld returned 1 exit status
Azure_Kinect_ROS_Driver/CMakeFiles/azure_kinect_ros_driver_node.dir/build.make:186: recipe for target '/home/sgrasp_robot/catkin_ws/devel/lib/azure_kinect_ros_driver/node' failed
make[2]: *** [/home/sgrasp_robot/catkin_ws/devel/lib/azure_kinect_ros_driver/node] Error 1
CMakeFiles/Makefile2:10401: recipe for target 'Azure_Kinect_ROS_Driver/CMakeFiles/azure_kinect_ros_driver_node.dir/all' failed
make[1]: *** [Azure_Kinect_ROS_Driver/CMakeFiles/azure_kinect_ros_driver_node.dir/all] Error 2
Makefile:145: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j16 -l16" failed
```

I think this is related to #127 then I tried to fix but I still get this error.
Does anyone know how to fix this bug ??
This error is caused by here.

```cpp
k4a_result_t K4AROSDevice::renderDepthToROS(sensor_msgs::ImagePtr& depth_image, k4a::image& k4a_depth_frame)
{
  cv::Mat depth_frame_buffer_mat(k4a_depth_frame.get_height_pixels(), k4a_depth_frame.get_width_pixels(), CV_16UC1,
                                 k4a_depth_frame.get_buffer());
// HERE //////////////////////////////////////////////////////////////////////////////////////////////////////////
  cv::Mat new_image(k4a_depth_frame.get_height_pixels(), k4a_depth_frame.get_width_pixels(), CV_32FC1);  

  depth_frame_buffer_mat.convertTo(new_image, CV_32FC1, 1.0 / 1000.0f);

  depth_image =
      cv_bridge::CvImage(std_msgs::Header(), sensor_msgs::image_encodings::TYPE_32FC1, new_image).toImageMsg();

  return K4A_RESULT_SUCCEEDED;
}

k4a_result_t K4AROSDevice::getIrFrame(const k4a::capture& capture, sensor_msgs::ImagePtr& ir_image)
{
  k4a::image k4a_ir_frame = capture.get_ir_image();

  if (!k4a_ir_frame)
  {
    ROS_ERROR("Cannot render IR frame: no frame");
    return K4A_RESULT_FAILED;
  }

  return renderIrToROS(ir_image, k4a_ir_frame);
}

k4a_result_t K4AROSDevice::renderIrToROS(sensor_msgs::ImagePtr& ir_image, k4a::image& k4a_ir_frame)
{
  cv::Mat ir_buffer_mat(k4a_ir_frame.get_height_pixels(), k4a_ir_frame.get_width_pixels(), CV_16UC1,
                        k4a_ir_frame.get_buffer());

  // Rescale the image to mono8 for visualization and usage for visual(-inertial) odometry.
  if (params_.rescale_ir_to_mono8)
  {
// HERE ///////////////////////////////////////////////////////////////////////////////////////////////////
    cv::Mat new_image(k4a_ir_frame.get_height_pixels(), k4a_ir_frame.get_width_pixels(), CV_8UC1);
    // Use a scaling factor to re-scale the image. If using the illuminators, a value of 1 is appropriate.
    // If using PASSIVE_IR, then a value of 10 is more appropriate; k4aviewer does a similar conversion.
    ir_buffer_mat.convertTo(new_image, CV_8UC1, params_.ir_mono8_scaling_factor);
    ir_image = cv_bridge::CvImage(std_msgs::Header(), sensor_msgs::image_encodings::MONO8, new_image).toImageMsg();
  }
  else
  {
    ir_image = cv_bridge::CvImage(std_msgs::Header(), sensor_msgs::image_encodings::MONO16, ir_buffer_mat).toImageMsg();
  }

  return K4A_RESULT_SUCCEEDED;
}
```

(I know if we use opencv 3.2.0 we can build but opencv3.2.0 is a little bit old and it's difficult to use with other libraries)

**To Reproduce**

Steps to reproduce the behavior:
1. Install Opencv3.4.13 from source
2. catkin_make

**Expected behavior**

Build success

**Logs**

**Screenshots**

**Desktop (please complete the following information):**
 - Ubuntu 18.04 (Docker)
 - opencv 3.4.13

**Additional context**

ooeygui, 2021-04-27:
We plan to support later OpenCV versions with Noetic or Foxy with 20.04 (allowing us to use later Body tracking SDK & newer Azure SDK)

Shunichi09, 2021-04-28:
Thank you for your reply. Ok. I'm looking forward to your new release.

closed 2021-05-02 by Shunichi09