microsoft/Azure_Kinect_ROS_Driver #193: Nvidia Jetson Xavier NX: catkin_make undefined reference to `cv::error(int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, char const*, char const*, int)'

vinesmsuic, 2021-05-22:
**Describe the bug**
I tried to install ros driver on 2 Nvidia Jetson Xavier NX machines but both yield the same error when executing `catkin_make`.

**To Reproduce**
Steps to reproduce the behavior:
1. Configure Microsoft's Package Repository and install k4atools
```
curl -sSL https://packages.microsoft.com/keys/microsoft.asc | sudo apt-key add -
sudo apt-add-repository https://packages.microsoft.com/ubuntu/18.04/multiarch/prod
sudo apt-get update
sudo apt install k4a-tools
sudo apt install libk4a1.4-dev
```
2. Inital the workspace
``` 
mkdir -p ~/AzureKinect_ws/src
cd ~/AzureKinect_ws/src
catkin_init_workspace
```
3.
```
git clone https://github.com/microsoft/Azure_Kinect_ROS_Driver.git
cd ~/AzureKinect_ws
catkin_make
```
4. See error

**Expected behavior**
successful catkin_make

**Logs**
```
robocon@robocon-desktop:~/AzureKinect_ws$ catkin_make
Base path: /home/robocon/AzureKinect_ws
Source space: /home/robocon/AzureKinect_ws/src
Build space: /home/robocon/AzureKinect_ws/build
Devel space: /home/robocon/AzureKinect_ws/devel
Install space: /home/robocon/AzureKinect_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/robocon/AzureKinect_ws/build"
####
####
#### Running command: "make -j6 -l6" in "/home/robocon/AzureKinect_ws/build"
####
[ 10%] Building CXX object Azure_Kinect_ROS_Driver/CMakeFiles/azure_kinect_ros_driver_nodelet.dir/src/k4a_ros_bridge_nodelet.cpp.o
[ 20%] Building CXX object Azure_Kinect_ROS_Driver/CMakeFiles/azure_kinect_ros_driver_node.dir/src/k4a_ros_device.cpp.o
[ 30%] Building CXX object Azure_Kinect_ROS_Driver/CMakeFiles/azure_kinect_ros_driver_nodelet.dir/src/k4a_ros_device.cpp.o
[ 40%] Building CXX object Azure_Kinect_ROS_Driver/CMakeFiles/azure_kinect_ros_driver_nodelet.dir/src/k4a_ros_device_params.cpp.o
[ 50%] Building CXX object Azure_Kinect_ROS_Driver/CMakeFiles/azure_kinect_ros_driver_node.dir/src/k4a_ros_bridge_node.cpp.o
[ 60%] Building CXX object Azure_Kinect_ROS_Driver/CMakeFiles/azure_kinect_ros_driver_node.dir/src/k4a_ros_device_params.cpp.o
[ 70%] Building CXX object Azure_Kinect_ROS_Driver/CMakeFiles/azure_kinect_ros_driver_nodelet.dir/src/k4a_calibration_transform_data.cpp.o
[ 80%] Building CXX object Azure_Kinect_ROS_Driver/CMakeFiles/azure_kinect_ros_driver_node.dir/src/k4a_calibration_transform_data.cpp.o
[ 90%] Linking CXX executable /home/robocon/AzureKinect_ws/devel/lib/azure_kinect_ros_driver/node
[100%] Linking CXX shared library /home/robocon/AzureKinect_ws/devel/lib/libazure_kinect_ros_driver_nodelet.so
[100%] Built target azure_kinect_ros_driver_nodelet
CMakeFiles/azure_kinect_ros_driver_node.dir/src/k4a_ros_device.cpp.o: In function `cv::Mat::Mat(int, int, int, void*, unsigned long)':
k4a_ros_device.cpp:(.text._ZN2cv3MatC2EiiiPvm[_ZN2cv3MatC5EiiiPvm]+0x150): undefined reference to `cv::error(int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, char const*, char const*, int)'
k4a_ros_device.cpp:(.text._ZN2cv3MatC2EiiiPvm[_ZN2cv3MatC5EiiiPvm]+0x244): undefined reference to `cv::error(int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, char const*, char const*, int)'
k4a_ros_device.cpp:(.text._ZN2cv3MatC2EiiiPvm[_ZN2cv3MatC5EiiiPvm]+0x2cc): undefined reference to `cv::Mat::updateContinuityFlag()'
collect2: error: ld returned 1 exit status
Azure_Kinect_ROS_Driver/CMakeFiles/azure_kinect_ros_driver_node.dir/build.make:214: recipe for target '/home/robocon/AzureKinect_ws/devel/lib/azure_kinect_ros_driver/node' failed
make[2]: *** [/home/robocon/AzureKinect_ws/devel/lib/azure_kinect_ros_driver/node] Error 1
CMakeFiles/Makefile2:519: recipe for target 'Azure_Kinect_ROS_Driver/CMakeFiles/azure_kinect_ros_driver_node.dir/all' failed
make[1]: *** [Azure_Kinect_ROS_Driver/CMakeFiles/azure_kinect_ros_driver_node.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j6 -l6" failed
```

**Desktop (please complete the following information):**
Device: Nvidia Jetson Xavier NX, Azure Kinect
OS: Ubuntu 18.04
ROS Version: ROS Melodic
K4A SDK Version: 1.4.0

**Additional context**
Tried on two Nvidia Jetson Xavier NX devices, both yield the same error. It seems to be a linker issue of OpenCV but I have no clue how to fix it.

ooeygui, 2021-05-27:
Thank you for reporting.
Jetpack includes a different version of OpenCV than ROS Melodic traditionally expects, which causes some issues when building projects that traditionally target Melodic. 
To support this scenario, the code needs to be updated to detect the OpenCV version difference - not just the distro difference. 

Pull requests to fix this issue are appreciated.

vinesmsuic, 2021-06-02:
In case people encountered the same problem as me,
I found a hack to solve this error.

add an extra `vision_opencv` package in `src` folder then `catkin_make` again
```
git clone https://github.com/MartinNievas/vision_opencv
```

ooeygui, 2021-06-02:
@vinesmsuic Thank you for following up and posting the workaround!