microsoft/Azure_Kinect_ROS_Driver #195: Body Tracking Data - Joint Rotation switching between positive and negative values

AswinkarthikeyenAK, 2021-06-04:
Hi, 
I started the body tracking and then captured the rotation of the pelvis joint and noticed that the data was not consistent.
Can someone kindly let me know why the data of the pelvis joint data obtained from the camera is not consistent with the rotation?  
[image]
Is there a way to fix this?
<arg name="body_tracking_smoothing_factor"  default="1" /> (Does this have any effect?)

Thanks

ooeygui, 2021-06-04:
@qm13 do you have any input on this?

zchenpds, 2021-06-04:
This is expected. You converted the quaternion to Euler angles and the yaw angle is wrapped to [-pi, pi), causing the discontinuity you observe in the yaw angle. The quaternion still changes continuously with time.

ooeygui, 2021-06-04:
@zhenpds Thank you for your input here!

closed 2021-06-04 by ooeygui