microsoft/Azure_Kinect_ROS_Driver #196: Question about Accessing .mkv files

wangmiaowei, 2021-06-21:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

wangmiaowei, 2021-06-21:
Well, I do not bring my Azure Kinect device now. But I want to process the .mkv file. How can I use the node to access my preoutputted .mkv file and convert it to point cloud?

ooeygui, 2021-06-21:
@wangmiaowei The Azure Kinect ROS node does not have this capability. The most accurate way to accomplish what you are looking for is to capture the output from the Azure Kinect to a ROSBag; then you can play it back later as if it came from the camera.

closed 2021-06-21 by ooeygui

zchenpds, 2021-06-21:
Thanks to @RoseFlunder, you may play back a .mkv file by starting `driver.launch` with the following argument specified.
https://github.com/microsoft/Azure_Kinect_ROS_Driver/blob/cac73214020db1298decb6ad0d22aac45863a531/launch/driver.launch#L39

ooeygui, 2021-06-21:
@zchenpds Wow, I totally forgot about that. Thank you for responding.

wangmiaowei, 2021-06-22:
Well, thanks for your suggestion.However, there are some differences with what I have expected. I want a end to end pipeline which  can output point clouds when I play .mkv file in this Azure Kinect ROS system in a real time.