microsoft/Azure_Kinect_ROS_Driver #198: Windows build failing

DyumanAditya, 2021-07-02:
Hi everyone,
I've installed the Azure-Kinect-Sensor-SDK for Windows 10 as instructed [here](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/usage.md#msis). But when I make a `catkin_ws`, clone the Azure_Kinect_ROS_Driver, and run `catkin_make --force-cmake`, I get the following failure message:

```
C:\Users\Dyuman\catkin_ws>catkin_make --force-cmake
Base path: C:\Users\Dyuman\catkin_ws
Source space: C:\Users\Dyuman\catkin_ws\src
Build space: C:\Users\Dyuman\catkin_ws\build
Devel space: C:\Users\Dyuman\catkin_ws\devel
Install space: C:\Users\Dyuman\catkin_ws\install
####
#### Running command: "cmake C:\Users\Dyuman\catkin_ws\src -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCATKIN_DEVEL_PREFIX=C:\Users\Dyuman\catkin_ws\devel -DCMAKE_INSTALL_PREFIX=C:\Users\Dyuman\catkin_ws\install -G Ninja" in "C:\Users\Dyuman\catkin_ws\build"
####
-- The C compiler identification is MSVC 19.28.29915.0
-- The CXX compiler identification is MSVC 19.28.29915.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.28.29910/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.28.29910/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: C:/Users/Dyuman/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: C:\opt\ros\noetic\x64\tools\vcpkg\installed\x64-windows;c:\opt\ros\noetic\x64
-- This workspace overlays: c:/opt/ros/noetic/x64
-- Found PythonInterp: C:/opt/ros/noetic/x64/python.exe (found suitable version "3.8.3", minimum required is "3")
-- Using PYTHON_EXECUTABLE: C:/opt/ros/noetic/x64/python.exe
-- Using default Python package layout
-- Found PY_em: C:\opt\ros\noetic\x64\lib\site-packages\em.py
-- Using empy: C:/opt/ros/noetic/x64/lib/site-packages/em.py
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: C:/Users/Dyuman/catkin_ws/build/test_results
-- Found gtest: gtests will be built
-- Using Python nosetests: C:/opt/ros/noetic/x64/Scripts/nosetests.exe
-- catkin 0.8.9
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 1 packages in topological order:
-- ~~  - azure_kinect_ros_driver
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'azure_kinect_ros_driver'
-- ==> add_subdirectory(Azure_Kinect_ROS_Driver)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
Finding K4A SDK binaries
!!! Body Tracking SDK not found: body tracking features will not be available !!!
K4A Libs: k4a::k4a;k4a::k4arecord
K4A DLLs: C:/Program Files/Azure Kinect SDK v1.4.1/sdk/windows-desktop/amd64/release/bin/k4a.dll;C:/Program Files/Azure Kinect SDK v1.4.1/sdk/windows-desktop/amd64/release/bin/depthengine_2_0.dll;C:/Program Files/Azure Kinect SDK v1.4.1/sdk/windows-desktop/amd64/release/bin/k4arecord.dll
K4A Install Needed: TRUE
-- Copied dll from k4a.dll to C:/Users/Dyuman/catkin_ws/devel/lib/azure_kinect_ros_driver
-- Copied dll from k4a.dll to C:/Users/Dyuman/catkin_ws/devel/lib
-- Copied dll from depthengine_2_0.dll to C:/Users/Dyuman/catkin_ws/devel/lib/azure_kinect_ros_driver
-- Copied dll from depthengine_2_0.dll to C:/Users/Dyuman/catkin_ws/devel/lib
-- Copied dll from k4arecord.dll to C:/Users/Dyuman/catkin_ws/devel/lib/azure_kinect_ros_driver
-- Copied dll from k4arecord.dll to C:/Users/Dyuman/catkin_ws/devel/lib
Installing K4A SDK to binary output folder
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/Dyuman/catkin_ws/build
####
#### Running command: "ninja -j8 -l8" in "C:\Users\Dyuman\catkin_ws\build"
####
ninja: error: 'C:/Program Files (x86)/Windows Kits/10/Lib/10.0.18362.0/um/x64/RpcRT4.Lib', needed by 'C:/Users/Dyuman/catkin_ws/devel/bin/azure_kinect_ros_driver_nodelet.dll', missing and no known rule to make it
Invoking "ninja -j8 -l8" failed
```

The Lib file `RpcRT4`  is present in the mentioned directory, so how can I fix this?

ooeygui, 2021-07-04:
@DyumanAditya Thanks for the report, I'd like to get to the bottom of this.
I've attempted to build with current tip of tree for the melodic-devel branch builds on Melodic and Noetic for the current release of the Windows builds for both melodic and noetic, which succeeded. 
Can you try removing the `devel` and `build` directories in the catkin workspace and rebuild?

DyumanAditya, 2021-07-05:
Thank you for your reply @ooeygui, I tried removing the devel and build directories and rebuilt, but received the same error.
I've been working on the default `melodic` branch from [here](https://github.com/microsoft/Azure_Kinect_ROS_Driver/tree/melodic)

ooeygui, 2021-07-06:
It looks like I was behind on VS. I'm updating now.

ooeygui, 2022-04-19:
This has been corrected

closed 2022-04-19 by ooeygui